espressif/esp-idf #4151: The LICENCE document is incomplete.

nevercast, 2019-10-04:
The LICENCE document at the root of this repository is only a copypaste of the Apache licence without all the fields filled in. There is no licencer nor a year provided. This licence is essentially incomplete.

Since I'm providing a full intellectual property / ownership document to investors I need to know exactly which entity owns which parts of our codebase. I would assume that ESP-IDF is owned and controlled by Espressif. But I don't see that in the LICENCE document.

Please ensure the LICENCE is complete at the earliest convienence, just to keep lawyers happy. 

Kind regards.

igrr, 2019-10-04:
@nevercast please refer to https://docs.espressif.com/projects/esp-idf/en/latest/COPYRIGHT.html, it contains the copyright statement and the license grant.

nevercast, 2019-10-04:
Thank you, I will refer to that URL in my document.

Alvin1Zhang, 2019-10-08:
@nevercast Thanks for reporting, feel free to reopen if you still have the issue, thanks.

closed 2019-10-08 by Alvin1Zhang