espressif/esp-idf #4152: Docker to build, macOS to flash error

cchaz003, 2019-10-04:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-369-g4dac7c7df
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [macOS]
- Power Supply:         [USB]

## Problem Description

I'm attempting to use docker the built-in IDF docker container to standardize my build environment across systems however I'm running into issues flashing via macOS after compiling on docker. I would use the docker image to flash directly however macOS and docker don't play well with usb support. The issue is after successfully compiling via docker I run "idf.py flash" and get an error related to cmake not being found (referencing a path from the docker container which is different from the cmake path on mac). I'm very new to all of this so this may be a dumb question but I'm wondering why cmake is needed to flash a fully compiled program? Once it is compiled in docker shouldn't idf.py be able to flash without dealing with the build system utilities? 

### Expected Behavior
Flashing of compiled project to ESP32 module

### Actual Behavior
error generated from running "idf.py flash" locally on macOS (*** inserted to hide personal info):

`Adding "flash"'s dependency "all" to list of commands with default set of options.
Executing action: all (aliases: build)
Running ninja in directory /***/***/***/***/***/***/build
Executing "ninja all"...
ninja: warning: bad deps log signature or version; starting over
[0/1] Re-running CMake...
/bin/sh: /usr/bin/cmake: No such file or directory
FAILED: build.ninja 
/usr/bin/cmake -H /***/***/***/***/***/***/ -B /***/***/***/***/***/***/build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1
`
### Steps to reproduce
on macOS
1. compile any example program via the latest docker image (docker run --rm -it -v $PWD:$PWD -w $PWD espressif/idf idf.py build)
2. run "idf.py flash"

igrr, 2019-10-04:
Hi @cchaz003, the issue is that `idf.py` flash is designed to also make sure that the binary is up to date, and invokes `idf.py build` to check that. Internally this calls CMake, which fails because the tool locations saved in CMakeCache.txt point to the files inside the container.

If the only thing you need is to flash the binary, you can call esptool.py directly, passing it the argument files generated by the build system: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#flash-arguments.

One other option, if you'd like to develop inside the container, is to run the rfc2217 server on the host (https://pyserial.readthedocs.io/en/latest/examples.html#single-port-tcp-ip-serial-bridge-rfc-2217) and allow the docker container to connect to the server running on the host. Then in the container you can set `ESPPORT=rfc2217://host:port` and call idf.py. Note that this may not work in some cases, due to the timing issues between esptool.py and the serial port driver. I am mentioning this approach just to lay out all the options, keep in mind that we aren't actively supporting or testing it.

cchaz003, 2019-10-04:
Ahh, ok thank you so much! I figured that was more or less how it worked but figured i'd ask why anyways. 

I did see that it could be done using esptool but I really like the idf.py implementation. I just setup the rfc2217 server and it's working great (and I think makes the most sense for my current situation). If things start not working i'll fall back on using esptool though.

Thanks for the quick response!

closed 2019-10-04 by cchaz003