espressif/esp-idf #4153: How to give different colours (HEX code of RGB) using PCA9685 and ESP32

ROYMURLI, 2019-10-04:
Hi,
I am using PCA9685 for RGB LED. I want different colours by using different HEX code.

In my understanding:-
setPWM(LED1, 0, 4096)  (means the pin is off and no light is there)
setPWM(LED1,4096,0) (means the pin os on and light will be there)

Now,
If I want red light then
setPWM(LED1,4096,0)
setPWM(LED2,0,4096)
setPWM(LED3,0,4096)  because the HEX value is FF0000 of RGB 

So I think if I need any colour with HEX code (A52A2A) in decimal (165,42,42) for brown colour
so should I do like this-
setPWM(LED1,4096*165/256,0)
setPWM(LED2,4096*42/256,0)
setPWM(LED3,4096*42/256,0)

I am not getting the proper result by using this. Can someone tell me how to generate any colour by using HEX code and setPWM function.

In the same way I came across a function:

esp_err_t fade_all_up(void)
{
	//esp_err_t ret;
	for (uint16_t i = 0; i < 128; i++)
	{
		for (uint8_t pin=0; pin<15; )//pin++)
		{
			// fade up
			setPWM(LEDs[pin], 32+32*i-1, 0);//pwmTable[i]);
			setPWM(LEDs[pin+1], 32+32*i-1, 0);//4096, 0);
			setPWM(LEDs[pin+2], 4096, 0);
			pin+=3;
		}

		vTaskDelay(10/portTICK_PERIOD_MS);
	}

	return ESP_OK; //ret;
}

this function lights the LED in dark yellow coulor slowly. I want to know how can I use different HEX code to light up any colour in same manner and same intensity.

ROYMURLI, 2019-10-09:
Can anyone help me to understand this?

ROYMURLI, 2019-10-11:
Hi everyone
Can anyone suggest me regarding this topic?
@Alvin1Zhang  @costaud  @blueMoodBHD

costaud, 2020-01-15:
"setPWM" is not the API provided by ESP-IDF. Are you using arduino platform, or ? Have you read the documentation of setPWM API ?

I think this is related to the LEDs you are using. if setPWM can set proper PWM waveforms with correct duty cycle and frequency, then this is not related to the PWM chip and esp32 controller.

ROYMURLI, 2020-01-16:
Thanks for your comment. I have resolved that issue.

closed 2020-01-16 by ROYMURLI