espressif/esp-idf #4154: Possible memory leak (newlib or cJSON?)

no1seman, 2019-10-04:
## Environment
- Development Kit:  ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used: ESP32-D0WD
- IDF version: v4.1-dev-369-g4dac7c7df
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-linux-amd64
- Operating System: Linux
- Power Supply:  USB|external 5V&Battery

## Problem Description
When using cJSON_PrintUnformatted got memory leak, possibly in sprintf()

### Expected Behavior
No any leaks
### Actual Behavior
16 bytes, 132 bytes, 28 bytes and 24 bytes = 200 bytes leaked (see log belog)
1680 bytes - it's Ok, it's actually JSON string itself and I free it later in code.

Problem seems to be in cJSON print_number() function, but can't understand whats wrong, or in newlib sprintf 
### Steps to repropduce

### Code to reproduce this issue
settings_json_root - pointer to root cJSON struct
```c
    heap_trace_init_standalone(trace_record, NUM_RECORDS);
    heap_trace_start(HEAP_TRACE_LEAKS);
    char *settings_json = cJSON_PrintUnformatted(settings_json_root);
    heap_trace_stop();
    heap_trace_dump();
```
## Debug Logs

```
5 allocations trace (1000 entry buffer)
16 bytes (@ 0x3ffd072c) allocated CPU 0 ccount 0x059d68f8 caller 0x400e76fe:0x400dd3a1:0x400db1e0:0x400d8f07:0x400d9097:0x400d9097:0x400d9097:0x400d9141:0x400d96ac:0x400d5b5c
0x400e76fe: _dtoa_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/dtoa.c:234 (discriminator 1)

0x400dd3a1: cvt at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1877
 (inlined by) _svfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1314

0x400db1e0: sprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/sprintf.c:620

0x400d8f07: print_number at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:510
 (inlined by) print_value at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1316

0x400d9097: print_object at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1681
 (inlined by) print_value at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1343

0x400d9097: print_object at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1681
 (inlined by) print_value at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1343

0x400d9097: print_object at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1681
 (inlined by) print_value at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1343

0x400d9141: print at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1116

0x400d96ac: cJSON_PrintUnformatted at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1170

0x400d5b5c: settings_to_json at /home/user/esp/beacon.new/build/../components/settings/settings_property.c:922

132 bytes (@ 0x3ffd0740) allocated CPU 0 ccount 0x059da260 caller 0x40086529:0x400e8b3c:0x400e91a4:0x400e77bc:0x400dd3a1:0x400db1e0:0x400d8f07:0x400d9097:0x400d9097:0x400d9097
0x40086529: _calloc_r at /home/user/esp/esp-mdf/esp-idf/components/newlib/heap.c:73

0x400e8b3c: _Balloc at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/mprec.c:106

0x400e91a4: __d2b at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/mprec.c:784

0x400e77bc: _dtoa_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/dtoa.c:287

0x400dd3a1: cvt at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1877
 (inlined by) _svfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1314

0x400db1e0: sprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/sprintf.c:620

0x400d8f07: print_number at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:510
 (inlined by) print_value at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1316

0x400d9097: print_object at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1681
 (inlined by) print_value at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1343

0x400d9097: print_object at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1681
 (inlined by) print_value at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1343

0x400d9097: print_object at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1681
 (inlined by) print_value at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1343

28 bytes (@ 0x3ffd07c8) allocated CPU 0 ccount 0x059db580 caller 0x40086529:0x400e8b68:0x400e91a4:0x400e77bc:0x400dd3a1:0x400db1e0:0x400d8f07:0x400d9097:0x400d9097:0x400d9097
0x40086529: _calloc_r at /home/user/esp/esp-mdf/esp-idf/components/newlib/heap.c:73

0x400e8b68: _Balloc at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/mprec.c:127

0x400e91a4: __d2b at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/mprec.c:784

0x400e77bc: _dtoa_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/dtoa.c:287

0x400dd3a1: cvt at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1877
 (inlined by) _svfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1314

0x400db1e0: sprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/sprintf.c:620

0x400d8f07: print_number at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:510
 (inlined by) print_value at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1316

0x400d9097: print_object at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1681
 (inlined by) print_value at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1343

0x400d9097: print_object at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1681
 (inlined by) print_value at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1343

0x400d9097: print_object at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1681
 (inlined by) print_value at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1343

24 bytes (@ 0x3ffd07e8) allocated CPU 0 ccount 0x059e4080 caller 0x40086529:0x400e8b68:0x400e79f1:0x400dd3a1:0x400db1e0:0x400d8f07:0x400d9097:0x400d9097:0x400d9097:0x400d9141
0x40086529: _calloc_r at /home/user/esp/esp-mdf/esp-idf/components/newlib/heap.c:73

0x400e8b68: _Balloc at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/mprec.c:127

0x400e79f1: _dtoa_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/dtoa.c:427

0x400dd3a1: cvt at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1877
 (inlined by) _svfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1314

0x400db1e0: sprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/sprintf.c:620

0x400d8f07: print_number at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:510
 (inlined by) print_value at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1316

0x400d9097: print_object at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1681
 (inlined by) print_value at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1343

0x400d9097: print_object at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1681
 (inlined by) print_value at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1343

0x400d9097: print_object at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1681
 (inlined by) print_value at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1343

0x400d9141: print at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1116

1680 bytes (@ 0x3ffd0804) allocated CPU 0 ccount 0x05b11b64 caller 0x400d9155:0x400d96ac:0x400d5b5c:0x400d32d8:0x400d10a1:0x40000000:
0x400d9155: print at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1125

0x400d96ac: cJSON_PrintUnformatted at /home/user/esp/esp-mdf/esp-idf/components/json/cJSON/cJSON.c:1170

0x400d5b5c: settings_to_json at /home/user/esp/beacon.new/build/../components/settings/settings_property.c:922

0x400d32d8: app_main at /home/user/esp/beacon.new/build/../main/main.c:33

0x400d10a1: main_task at /home/user/esp/esp-mdf/esp-idf/components/esp32/cpu_start.c:570

1880 bytes 'leaked' in trace (5 allocations)
total allocations 9 total frees 4

```

no1seman, 2019-10-04:
Finally got simple example to reproduce:
```c
#include <esp_types.h>
#include "freertos/FreeRTOS.h"
#include "esp_heap_trace.h"

#define NUM_RECORDS 1000
static heap_trace_record_t trace_record[NUM_RECORDS]; // This buffer must be in internal RAM

void app_main(void)
{
    heap_trace_init_standalone(trace_record, NUM_RECORDS);
    heap_trace_start(HEAP_TRACE_LEAKS);
    double test = 23;
    printf("%g\n", test);
    heap_trace_stop();
    heap_trace_dump();
}
```
Does this memory beeing allocated once then will be reused by _dtoa_r/_dtoa it on every *printf calls?

Log:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6996
load:0x40078000,len:14204
load:0x40080400,len:4340
entry 0x400806e8
I (71) boot: Chip Revision: 1
I (72) boot_comm: mismatch chip revision, expect 1, found 0
I (38) boot: ESP-IDF v4.1-dev-369-g4dac7c7df-dirty 2nd stage bootloader
I (39) boot: compile time 13:43:03
I (49) boot: Enabling RNG early entropy source...
I (49) boot: SPI Speed      : 40MHz
I (49) boot: SPI Mode       : DIO
I (53) boot: SPI Flash Size : 4MB
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: mismatch chip revision, expect 1, found 0
I (101) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x05474 ( 21620) map
I (118) esp_image: segment 1: paddr=0x0001549c vaddr=0x3ffb0000 size=0x0206c (  8300) load
I (122) esp_image: segment 2: paddr=0x00017510 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/user/esp/esp-mdf/esp-idf/components/freertos/xtensa_vectors.S:1778

I (128) esp_image: segment 3: paddr=0x00017918 vaddr=0x40080400 size=0x086f8 ( 34552) load
I (151) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x12438 ( 74808) map
0x400d0018: _stext at ??:?

I (179) esp_image: segment 5: paddr=0x00032458 vaddr=0x40088af8 size=0x014d8 (  5336) load
0x40088af8: portENTER_CRITICAL_NESTED at /home/user/esp/esp-mdf/esp-idf/components/freertos/include/freertos/portmacro.h:330
 (inlined by) vTaskEnterCritical at /home/user/esp/esp-mdf/esp-idf/components/freertos/tasks.c:4196

I (188) boot: Loaded app from partition at offset 0x10000
I (188) boot: Disabling RNG early entropy source...
I (188) cpu_start: Pro cpu up.
I (192) cpu_start: Application information:
I (197) cpu_start: Project name:     hello-world
I (202) cpu_start: App version:      1
I (206) cpu_start: Compile time:     Oct  4 2019 13:43:05
I (212) cpu_start: ELF file SHA256:  35d7fdcc691fb111...
I (218) cpu_start: ESP-IDF:          v4.1-dev-369-g4dac7c7df-dirty
I (225) cpu_start: Starting app cpu, entry point is 0x40081028
0x40081028: call_start_cpu1 at /home/user/esp/esp-mdf/esp-idf/components/esp32/cpu_start.c:276
[... 100 lines cut ...]
0x400d53f8: printf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/printf.c:59

0x400d3659: app_main at /home/user/esp/printf_bug/build/../main/hello_world_main.c:15

0x400d0a4f: main_task at /home/user/esp/esp-mdf/esp-idf/components/esp32/cpu_start.c:570

0x40087774: vPortTaskWrapper at /home/user/esp/esp-mdf/esp-idf/components/freertos/port.c:145

84 bytes (@ 0x3ffcd27c) allocated CPU 0 ccount 0x045047f0 caller 0x40087d8a:0x40087fe7:0x40082ef9:0x40082f23:0x40083070:0x400d324e:0x400d21b8:0x4000bd86:
0x40087d8a: xQueueGenericCreate at /home/user/esp/esp-mdf/esp-idf/components/freertos/queue.c:392

0x40087fe7: xQueueCreateMutex at /home/user/esp/esp-mdf/esp-idf/components/freertos/queue.c:499

0x40082ef9: lock_init_generic at /home/user/esp/esp-mdf/esp-idf/components/newlib/locks.c:80

0x40082f23: lock_acquire_generic at /home/user/esp/esp-mdf/esp-idf/components/newlib/locks.c:135

0x40083070: _lock_acquire_recursive at /home/user/esp/esp-mdf/esp-idf/components/newlib/locks.c:172

0x400d324e: uart_write at /home/user/esp/esp-mdf/esp-idf/components/vfs/vfs_uart.c:196

0x400d21b8: esp_vfs_write at /home/user/esp/esp-mdf/esp-idf/components/vfs/vfs.c:420 (discriminator 4)

368 bytes 'leaked' in trace (6 allocations)
total allocations 6 total frees 0

no1seman, 2019-10-05:
Closing, found an answer that that was made by desingn to avoid memory fragmentation

closed 2019-10-05 by no1seman

zhanghuichina, 2020-03-13:
how to avoid? @no1seman

no1seman, 2020-03-13:
@zhanghuichina ,

short answer: no way

middle answer: do not use newlib *printf family functions, for example try to use jsmn,  it doesn't use *printf, but as for me it's a little bit more difficult to use

long answer: *printf family functions uses some allocations on first call and then they doesn't release buffers, but reuses it. This needs to avoid memory fragmentation when you intensively using *printf family functions from stdlib. You can easily check it by making simple test with calling sprintf in cycle.
Seems to me that lost of some hundreds of bytes is not a problem at all