espressif/esp-idf #4155: can not compile ulp example

kekemuyu, 2019-10-04:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      esp32-dev
- Kit version (for WroverKit/PicoKit/DevKitC): 
- Module or chip used:  [SP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): 
    v4.1-dev-369-g4dac7c7df
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Linux]
- Power Supply:         [external 5V]

## Problem Description

**when run "idf.py build" at  ~/esp/ulp folder,    the error log：**

`Executing action: all (aliases: build)
Running ninja in directory /home/kekemuyu/esp/ulp_adc/build
Executing "ninja all"...
[1/15] Performing build step for 'bootloader'
ninja: no work to do.
[2/13] Performing configure step for 'ulp_main'
FAILED: esp-idf/main/ulp_main-prefix/src/ulp_main-stamp/ulp_main-configure 
cd /home/kekemuyu/esp/ulp_adc/build/esp-idf/main/ulp_main && /usr/bin/cmake -DCMAKE_GENERATOR=Ninja -DCMAKE_TOOLCHAIN_FILE=/home/kekemuyu/esp/esp-idf/components/ulp/cmake/toolchain-ulp.cmake -DULP_S_SOURCES=/home/kekemuyu/esp/ulp_adc/main/ulp/adc.S -DULP_APP_NAME=ulp_main -DCOMPONENT_DIR=/home/kekemuyu/esp/ulp_adc/main "-DCOMPONENT_INCLUDES=/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/newlib/platform_include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/freertos/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/heap/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/log/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/soc/esp32/include;/home/kekemuyu/esp/esp-idf/components/soc/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/esp_rom/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/esp_common/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/xtensa/include;/home/kekemuyu/esp/esp-idf/components/xtensa/esp32/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/esp32/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/driver/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/esp_ringbuf/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/esp_event/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/tcpip_adapter/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/lwip/include/apps;/home/kekemuyu/esp/esp-idf/components/lwip/include/apps/sntp;/home/kekemuyu/esp/esp-idf/components/lwip/lwip/src/include;/home/kekemuyu/esp/esp-idf/components/lwip/port/esp32/include;/home/kekemuyu/esp/esp-idf/components/lwip/port/esp32/include/arch;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/vfs/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/esp_wifi/include;/home/kekemuyu/esp/esp-idf/components/esp_wifi/esp32/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/esp_eth/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/efuse/include;/home/kekemuyu/esp/esp-idf/components/efuse/esp32/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/app_trace/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/nvs_flash/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/spi_flash/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/mbedtls/port/include;/home/kekemuyu/esp/esp-idf/components/mbedtls/mbedtls/include;/home/kekemuyu/esp/ulp_adc/build/config;/home/kekemuyu/esp/esp-idf/components/ulp/include;/home/kekemuyu/esp/ulp_adc/build/config" -DIDF_PATH=/home/kekemuyu/esp/esp-idf -DSDKCONFIG=/home/kekemuyu/esp/ulp_adc/build/config/sdkconfig.h -GNinja /home/kekemuyu/esp/esp-idf/components/ulp/cmake && /usr/bin/cmake -E touch /home/kekemuyu/esp/ulp_adc/build/esp-idf/main/ulp_main-prefix/src/ulp_main-stamp/ulp_main-configure
-- The ASM compiler identification is unknown
-- Found assembler: esp32ulp-elf-as
CMake Error at CMakeLists.txt:4 (project):
  The CMAKE_ASM_COMPILER:

    esp32ulp-elf-as

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "/home/kekemuyu/esp/ulp_adc/build/esp-idf/main/ulp_main/CMakeFiles/CMakeOutput.log".
See also "/home/kekemuyu/esp/ulp_adc/build/esp-idf/main/ulp_main/CMakeFiles/CMakeError.log".
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
`

chegewara, 2019-10-04:
Have you installed this:
https://github.com/espressif/binutils-esp32ulp

https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/ulp-legacy.html#installing-the-toolchain

renzbagaporo, 2019-10-11:
Did you install the ULP toolchain?  https://github.com/espressif/binutils-esp32ulp

Make sure that this is in your path. To check you should be able to execute `esp32ulp-elf-as --version`.

```
user@pc:/esp-idf$ esp32ulp-elf-as  --version
GNU assembler (GNU Binutils) 2.28.51.20170517
Copyright (C) 2017 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or later.
This program has absolutely no warranty.
This assembler was configured for a target of `esp32ulp-elf'.
```

chegewara, 2019-10-12:
@renzbagaporo Initially i thought thats the case, but he is using CMake and according docs ULP should be already included in it.
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/ulp.html#installing-the-toolchain

kekemuyu, 2019-10-16:
> @renzbagaporo Initially i thought thats the case, but he is using CMake and according docs ULP should be already included in it.
> https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/ulp.html#installing-the-toolchain

that's right,I do it accord this docs

igrr, 2019-11-04:
@kekemuyu can you please run the command `esp32ulp-elf-as --version` as requested above by @renzbagaporo? Just to verify that the tool can be executed.

Also, are you using a 32-bit version of Linux, by any chance?

Alvin1Zhang, 2020-04-29:
@kekemuyu Thanks for reporting, will close due to short of feedback, feel free to reopen with more details. Thanks.

closed 2020-04-29 by Alvin1Zhang