espressif/esp-idf #4157: SystemView with AppTrace Lock does not work

drorgl, 2019-10-04:
**This issue could be related to PlatformIO build, see analysis**

## Environment
- Development Kit:      ESP32-DevKitC
- Kit version: v4
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 3.30300.190916
- Build System:        PlatformIO
- Compiler version 1.22.0-80-g6c4433a5
- Operating System:     Windows
- Power Supply:         USB
- JTAG: Segger J-Link

## Problem Description
Attempting to enable SystemView with ```CONFIG_ESP32_APPTRACE_LOCK_ENABLE``` results in the following errors:
```
(0) cpu_start: App cpu up.
(219) heap_init: Initializing. RAM available for dynamic allocation:
(226) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
...
(257) cpu_start: Pro cpu start user code
assertion "res == coreID || res == portMUX_FREE_VAL" failed: file "portmux_impl.inc.h", line 105, function: vPortCPUAcquireMutexIntsDisabledInternal
abort() was called at PC 0x400d32e3 on core 0
```
where this section repeats and eventually resets:
```
assertion "res == coreID || res == portMUX_FREE_VAL" failed: file 
"portmux_impl.inc.h", line 105, function: vPortCPUAcquireMutexIntsDisabledInternal
```

### Expected Behavior
SystemView should work as described in the documentation.

### Actual Behavior
SystemView can only be used when ```CONFIG_ESP32_APPTRACE_LOCK_ENABLE``` is ```0```

### Steps to repropduce
Create a new project with one endless loop task and the following settings:
```
#define CONFIG_ESP32_APPTRACE_ENABLE 1
#define CONFIG_ESP32_APPTRACE_LOCK_ENABLE 1
#define CONFIG_SYSVIEW_ENABLE 1

//extras:
#define CONFIG_SYSVIEW_MAX_TASKS 15
#define CONFIG_FREERTOS_GENERATE_RUN_TIME_STATS 1
#define CONFIG_FREERTOS_USE_STATS_FORMATTING_FUNCTIONS 1
#define CONFIG_FREERTOS_USE_TRACE_FACILITY 1
#define CONFIG_ESP32_APPTRACE_POSTMORTEM_FLUSH_TRAX_THRESH 0
#define CONFIG_ESP32_APPTRACE_ONPANIC_HOST_FLUSH_TMO 10000
#define CONFIG_ESP32_APPTRACE_POSTMORTEM_FLUSH_THRESH 1
#define CONFIG_ESP32_APPTRACE_PENDING_DATA_SIZE_MAX 1024
#define CONFIG_ESP32_APPTRACE_DEST_TRAX 1
#define CONFIG_ESP32_DEBUG_STUBS_ENABLE 1
#define CONFIG_SYSVIEW_TS_SOURCE_ESP_TIMER 1
#define CONFIG_ESP32_TRAX 1
#define CONFIG_ESP32_TRAX_TWOBANKS 1
```

### Analysis
At first it looked like the issue is in ```start_cpu0_default()``` where ```esp_apptrace_init()``` is not called at the correct location, but even after moving it above ```esp_timer_init()``` a different error started, I think the lock might not get released in the trax code so its causing other side effects, but I did not finish the analysis.

After disabling the AppTrace locks it seems to be working.

This issue could be related to the way PlatformIO builds esp-idf or a regression in AppTrace / SystemView implementation or inappropriate defines.

gerekon, 2019-10-08:
@drorgl `CONFIG_ESP32_APPTRACE_LOCK_ENABLE` is incompatible with SystemView tracing module and should be [disabled](https://github.com/espressif/esp-idf/blob/master/components/app_trace/Kconfig#L25-L29).
SystemView uses its own locking [mechanism](https://github.com/espressif/esp-idf/blob/master/components/app_trace/sys_view/Sample/Config/SEGGER_SYSVIEW_Config_FreeRTOS.c#L340-L353).

drorgl, 2019-10-08:
@gerekon  I was a bit suspicious when I saw SystemView locks, thanks for clearing it up!