espressif/esp-idf #4158: can.c driver does not define IRAM_ATTR on interrupt routines

arjanmels, 2019-10-04:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it):  v3.2.3
- Build System:         [platformio]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a5 5.2.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

Hangs in nvs routines when using can driver simultaneously. I think it is due to missing IRAM_ATTR on the can interrupt routines (can_intr_handler_main etc.)

### Expected Behavior

Can use nvs and can simultaneously without hangs

### Actual Behavior

Occasional hangs when using nvs and can simultaneously

### Steps to repropduce

use can library (can_receive) in task and in another task use nvs_set_... functions

### Code to reproduce this issue

Too extensive to include

## Debug Logs

N/A: hangs without crash (unless WDT is enabled, then you get a WDT)

## Other items if possible

BeckerMarkus, 2019-10-05:
Good point. I do not use ESP-IDF's CAN driver, but an other design based on [Thomas Barth](https://www.barth-dev.de/can-driver-esp32/)'s work for historical reasons. I always wondered, if `IRAM_ATTR` would be authoritative for the CAN ISR to work reliably but did not manage to find a clear answer by now. So I have a provision like that in the code:

```
/**
 * define ISR_IN_IRAM to allocate the interrupt service routine in IRAM.
 * By default this is disabled (defined as NOT_ISR_IN_IRAM).
 */
#define NOT_ISR_IN_IRAM

#ifdef ISR_IN_IRAM
static  IRAM_ATTR  void CAN_isr(void *arg_p){
#endif
#ifndef ISR_IN_IRAM
static void CAN_isr(void *arg_p){
#endif
```

Could you just give it a try, set `IRAM_ATTR` for the ISR and see if it solves the issue?

Best
Markus

Dazza0, 2019-10-10:
@arjanmels thanks for reporting this issue. Could you post the Task WDT error log you're getting. I want to see exactly which task is triggering the Task, Watchdog timeout.

I'm not sure if this is an `IRAM_ATTR` issue. Because none of the CAN driver functions are placed in IRAM, the CAN driver will not run whilst flash cache is disabled, but it should not crash either. The CAN peripheral may lose a few messages if NVS is disabled for too long, but it should not crash.

Could you check the on the duration of your NVS access (i.e., get the system time before and after the `nvs_set` operation).

arjanmels, 2019-10-11:
In the meantime I worked around it by using can_stop before the NVS accesses and can_start afterwards. (And the hardware I was working on is delivered to the customer, so I cannot do additional testing at the moment.)

I indeed noticed that a lot of routines are called form the interrupt handler.

To me it looks like a deadlock: CAN interrupt during the NVS access cannot be served because the interrupt routine is in flash, but also teh NVS routine cannot continue.

Dazza0, 2021-03-09:
CAN/TWAI driver's ISR supports the option of being placed in IRAM as of commit 4c57f50fe4abb53e2b48e9237f3037d8bd79aff2 (v4.3), a0bb9b2f57ccd8cec5d258a9a8bad0fef3dfefcf (v4.2), backport for v4.1 branch is still pending. This feature can be enabled in menuconfig AND providing the `ESP_INTR_FLAG_IRAM` in the driver configuration structure.

closed 2021-03-09 by Dazza0