espressif/esp-idf #4159: HSP

howroyd, 2019-10-04:
Linked to [issue 1118](https://github.com/espressif/esp-idf/issues/1118)  and more of a "how do I..." rather than an issue.

I have an ESP32 dev kit and a standard Bluetooth speaker (with mic).  I can connect the ESP to my phone using HFP, seemingly with no issues, and receive SCO packets.

How can I use the bluetooth speaker/mic to send/receive audio?  I am assuming HSP but can't find any documentation on this.

Note; ultimately I do not need the phone to be the gateway, in fact I don't need HFP at all.  I am effectively setting up the ESP as a means to send I2S audio to a bluetooth speaker and pump out the bluetooth speakers microphone audio over another I2S line.

TIA

mywang-espressif, 2019-10-08:
Hi @howroyd ,
Bluetooth Hands-Free Profile defines two roles(configuration of devices): Audio Gateway(AG) and Hands-Free unit(HF). ESP32 Bluetooth host stack currently only supports HF, so it can connect to the phone which acts as AG.
As your bluetooth speaker also works as HF, it cannot work with the ESP32 currently.
HFP AG is under development on ESP32, and hopefully it can be released within a couple of months.

Your requirement is to transmit/receive audio stream with bluetooth speaker microphone. But Bluetooth audio link can only be set up according to the standard procedure defined in HFP which the standard bluetooth speaker follows.

As a gentle reminder, bluetooth (e)SCO link used in HFP to transmit audio has a limited bandwidth and audio quality. Typically, the audio stream is transcoded into CVSD or mSBC format through (e)SCO link, and finally the decoded audio stream is 8000Hz(for CVSD) and 16000Hz(for mSBC) sampled, 16-bit PCM stream. Please also check whether this quality is good enough for your application.

howroyd, 2019-10-08:
Thanks for the info.

The bandwidth limitations are fine for my application.

HSP is the preferred solution but I can't find any documentation on this. HFP is a fallback.

Is the esp32 capable of HSP now or in the future (how long until?).

Also, I am happy to assist HFP & HSP testing if you need help.

beckmx, 2019-10-21:
are there any examples for the HSP?

Wth-Esp, 2021-01-21:
Hi all,

On the code side, I think it's relatively easy to change HFP to HSP. Just change the service ID.

May you please try [enable_hsp.zip](https://github.com/espressif/esp-idf/files/5847574/enable_hsp.zip) on the master branch of IDF.

Just use the HFP APIs as HFP applications do to actually use HSP, in the meanwhile, the HFP demo comes into an HSP demo.

P.S. I haven't tried all commands in the demo.

Thanks

Wth-Esp, 2021-01-22:
I forget the SDP and `AT+CPKD` command, it's still HFP....

Wth-Esp, 2021-05-11:
Hi, all

As we estimate the feature, I'm sorry to say that we have no plan to support HSP on ESP32 in the predicted future. We think HFP can serve the market very well and there is no need to support HSP in the meanwhile.

Sorry for the late reply. 

Thanks

closed 2021-05-12 by tom-borcin

Qamada, 2022-06-06:
Hi @Wth-Esp @mywang-espressif 

> HFP AG is under development on ESP32, and hopefully it can be released within a couple of months.
Has espressif already released the library or protocol stack to implement ESP32 as HFP AG?

When I look for an example source code to follow HFP AG, I was unable to find it on the examples of espressif for ESP32.

Thanks

Wth-Esp, 2022-06-06:
Hi, @Qamada 

HFP AG is available on release/v4.1 and above version of IDF.

Thanks

Qamada, 2022-06-06:
> HFP AG is available on release/v4.1 and above version of IDF.

Thanks for the reply. I'm gonna try to use its HFP AG.