espressif/esp-idf #4161: xtensa-esp32-elf/bin/ld.exe: action.elf section `.iram0.text' will not fit in region `iram0_0_seg'

filzek, 2019-10-04:
Lastest master as of oct 10 2019.

After add the BLE to the project and call:
ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));
esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
ret = esp_bt_controller_init(&bt_cfg);

The error: 
xtensa-esp32-elf/bin/ld.exe: action.elf section `.iram0.text' will not fit in region `iram0_0_seg'
xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: IRAM0 segment data does not fit.
/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 15732 bytes

So, how can we fix this and find out where is the problem, so everything is running fine, but once add BLE we cant build anymore.

filzek, 2019-10-04:
We have removed all assignements and debugs and set to production with size optimization, the build runs but tasks stop from being created now. So, there is a problem when using BLE/BT, wifi, https, that breaks something we couldnt find as an error.

filzek, 2019-10-04:
E (118843) wl_flash: read(612): result = 0x00000101
E (118843) ff_diskio_spiflash: wl_read failed (257)
W (118843) vfs_fat_spiflash: f_mount failed (1)
E (118853) action: Failed to mount FATFS (ESP_FAIL)

now, a lot of error are being throw so far.

filzek, 2019-10-04:
W (144223) wifi: alloc eb len=24 type=3 fail, heap:4190312

W (144223) wifi: m f null

filzek, 2019-10-04:
now, udp and tcp corrupts everything, 
W (190712) wifi: alloc eb len=24 type=3 fail, heap:4189616
W (190712) wifi: m f null

filzek, 2019-10-04:
If set memory at spiram first, it crash worstly.

SPI RAM CONFIG:
(16384) Maximum malloc() size, in bytes, to always put in internal memory

if enable this:
[*] Try to allocate memories of WiFi and LWIP in SPIRAM firstly. If failed, allocate inte│ │ 

system just halt without any explanation.

-> I will try to erase sdkconfig and create a fresh default one, so, maybe there is something broken the building.

filzek, 2019-10-28:
Latest update solve partial building and show image image not fit, things are getting solved, seems that very soon all this matter could be closed.

negativekelvin, 2019-10-29:
https://github.com/espressif/esp-idf/issues/3592

closed 2019-10-31 by filzek