espressif/esp-idf #4162: Add color output to build output log (more readable errors and warnings)

AloyseTech, 2019-10-04:
## Environment

- Development Kit:      ESP32-PICO-Kit
- Kit version v4
- Module or chip used:  ESP32-PICO-D4
- IDF version v4.1-dev-256-g9f145ff16
- Build System:        CMake
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

The build process output a lot of text. It is sometimes difficult to spot errors and warnings.

### Expected Behavior

It could be great if the output could be colorized (red for errors, orange/yellow for warning etc), to make the build log more readable.

AloyseTech, 2019-10-04:
I've added `add_compile_options (-fdiagnostics-color=always)` to the main CMakeList.txt file of my project and set an environment variable `GCC_COLORS="error=01;31:warning=01;33:note=01;36:caret=01;32:locus=01:quote=01"` which give me some colored output. Is there any better way to do this?

renzbagaporo, 2019-10-14:
There is no official support for this in the sense that you can't pass an argument to `idf.py` to enable this. You can, however, skip adding `add_compile_options (-fdiagnostics-color=always)` to every project you work with by adding the -`fdiagnostics-color=always` to the environment variable EXTRA_CFLAGS and EXTRA_CXXFLAGS.

You still need to set the GCC_COLORS however for the colors you want the output to be.

esanai, 2019-11-22:
You can colorize the log with following code.
```
static const char *TAG = "TEST_TASK";
ESP_LOG_LEVEL(ESP_LOG_ERROR, TAG, "%d - Red log", 1);
ESP_LOG_LEVEL(ESP_LOG_WARN, TAG, "%d - Yello log", 2);
ESP_LOG_LEVEL(ESP_LOG_INFO, TAG, "%d - Green log", 3);
```
I found the way from the source code of [esp_log.h](https://github.com/espressif/esp-idf/blob/master/components/log/include/esp_log.h#L319)

renzbagaporo, 2019-11-22:
@esanai I think @AloyseTech was referring to colorizing the actual output build log, not the runtime logs.

esanai, 2019-11-23:
@renzbagaporo You are right. I'm sorry for my misunderstanding.

KaeLL, 2020-02-04:
> There is no official support for this in the sense that you can't pass an argument to `idf.py` to enable this. You can, however, skip adding `add_compile_options (-fdiagnostics-color=always)` to every project you work with OR by adding the -`fdiagnostics-color=always` to the environment variable EXTRA_CFLAGS and EXTRA_CXXFLAGS to have it be "permanent".
> 
> You still need to set the GCC_COLORS however for the colors you want the output to be.

Or the devs could actually fix this because, you know, it works with Make?
This issue needs to be reopened.

eflukx, 2020-05-13:
Add the parameters onto the environment, by executing:

```sh
export EXTRA_CFLAGS=-fdiagnostics-color=always
export GCC_COLORS="error=01;31:warning=01;35:note=01;36:range1=32:range2=34:locus=01:quote=01:path=01;36:fixit-insert=32:fixit-delete=31:diff-filename=01:diff-hunk=32:diff-delete=31:diff-insert=32:type-diff=01;32"
```

before my `idf.py build` command, but still only B&W output. Obviously doing something wrong.

Ps. Putting the some definitive handles here how to enable colour build-output would would surely help a lot of people, as this issue is where people land when searching for coloured build output.

hermlon, 2021-03-09:
For me the approach above of setting `EXTRA_CFLAGS` to `-fdiagnostics-color=always` did show colored output in `idf.py build` after running `idf.py fullclean` once.

Nodraak, 2021-04-27:
@renzbagaporo Please consider reopening this issue. I believe a colorized output should be the default in 2021.

wenbozzz, 2021-07-21:
Hi all, 
just want to leave a note here..
if you are using newer version of the `esp-idf`, such as v4.3, you can add this line to your project CMake file **after** the line `project(<project name>)`
```bash
idf_build_set_property(COMPILE_OPTIONS "-fdiagnostics-color=always" APPEND)
```

if you have other flag to set, you'll need to write them in separate line, e.g.
```bash
idf_build_set_property(COMPILE_OPTIONS "-Wno-error=class-memaccess" APPEND)
idf_build_set_property(COMPILE_OPTIONS "-fdiagnostics-color=always" APPEND)
```
since `idf_build_set_property(COMPILE_OPTIONS "-Wno-error=class-memaccess -fdiagnostics-color=always" APPEND)` didn't work for me.

ruipcunha, 2021-09-09:
@wenbozzz
I'm using IDF v4.3 and it works perfectly! Thanks a lot!

KaeLL, 2021-09-09:
> if you are using newer version of the `esp-idf`, such as v4.3, you can add this line to your project CMake file **after** the line `project(<project name>)`

IDF can and should set this by default, as there's no downside and it's still a (unnecessary) downgrade from Make.
@projectgus thoughts?

igrr, 2021-09-09:
@KaeLL Thanks for the ping. I think enabling `-fdiagnostics-color=always` in the project CMakeLists.txt has the side effect of adding the color escape sequences to build output for non-interactive builds. We plan to fix the underlying issues preventing the color output from being enabled by default in the console. The corresponding internal issue was already re-opened, but I forgot to re-open this one. It will be updated when someone starts working on this!

igrr, 2022-03-09:
Just an update, we haven't forgotten about this issue. Support for color diagnostics has just been merged into CMake (https://gitlab.kitware.com/cmake/cmake/-/merge_requests/6990) and we plan to upgrade the bundled CMake version as soon as the new version is released.

igrr, 2022-08-01:
Another update: CMake 3.24.0 [is about to be released](https://gitlab.kitware.com/cmake/cmake/-/milestones/102#tab-issues) and we now have a work-in-progress IDF branch which uses the new `CMAKE_COLOR_DIAGNOSTICS` feature introduced by CMake. Some work still needs to be done, but I expect the feature to be ready in a week or two, provided no issues are discovered with the new CMake release version.

igrr, 2022-09-02:
Color output from idf.py should now be available in the `master` branch, as long as CMake 3.24 or later is used. On Windows you can get it by running install.bat or install.ps1. On Linux or macOS it's possible to install it either system-wide, or for IDF only via `tools/idf_tools.py install cmake`.
Please give it a try, feedback is welcome!

closed 2022-11-16 by Alvin1Zhang