espressif/esp-idf #4164: Proper BLE init and deinit without memory leak

edubadia, 2019-10-05:
How do I can init and deinit BLE in BTDM mode, without memory leak?

I'm trying to develop a code which starts ble, allow someone to connect and disconnect, deinit ble and init ble again... repeatadly.

But I found out that I'm leaking RAM memory every time I init and deinit.

Can someone provide, the proper way to do it? As I'm finding some similar cases with similar issues, but I can't find the proper way to make it work.

init method:
  esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
  esp_bt_controller_init(&bt_cfg);
  esp_bt_controller_enable(ESP_BT_MODE_BTDM);
  esp_bluedroid_init();
  esp_bluedroid_enable();

deinit method:
  esp_bluedroid_disable();
  esp_bluedroid_deinit();
  esp_bt_controller_disable();
  esp_bt_controller_deinit();

gengyuchao, 2019-11-26:
Hello, I have tried to initialize and de-initialize the bt program many times, and I haven't found a memory leak. Here is the program I used.
```c

void start_ble_func()
{
    esp_err_t ret;

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    ret = esp_bt_controller_init(&bt_cfg);
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s initialize controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    ret = esp_bt_controller_enable(ESP_BT_MODE_BLE);
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s enable controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
    ret = esp_bluedroid_init();
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s init bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
    ret = esp_bluedroid_enable();
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s enable bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    ret = esp_ble_gatts_register_callback(gatts_event_handler);
    if (ret){
        ESP_LOGE(GATTS_TAG, "gatts register error, error code = %x", ret);
        return;
    }
    ret = esp_ble_gap_register_callback(gap_event_handler);
    if (ret){
        ESP_LOGE(GATTS_TAG, "gap register error, error code = %x", ret);
        return;
    }
    ret = esp_ble_gatts_app_register(PROFILE_A_APP_ID);
    if (ret){
        ESP_LOGE(GATTS_TAG, "gatts app register error, error code = %x", ret);
        return;
    }
    ret = esp_ble_gatts_app_register(PROFILE_B_APP_ID);
    if (ret){
        ESP_LOGE(GATTS_TAG, "gatts app register error, error code = %x", ret);
        return;
    }
    esp_err_t local_mtu_ret = esp_ble_gatt_set_local_mtu(500);
    if (local_mtu_ret){
        ESP_LOGE(GATTS_TAG, "set local  MTU failed, error code = %x", local_mtu_ret);
    }
}

void stop_ble_func()
{
    esp_err_t ret;

    ret = esp_bluedroid_disable();
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s esp_bluedroid_disable failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
    ret = esp_bluedroid_deinit();
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s esp_bluedroid_deinit failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
    ret = esp_bt_controller_disable();
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s esp_bt_controller_disable failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
    ret = esp_bt_controller_deinit();
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s esp_bt_controller_deinit failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
}

void app_main(void)
{
    esp_err_t ret;

    // Initialize NVS.
    ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );

    int free_heap_size = esp_get_free_heap_size();

    for(int i = 0;  i <10 ; i++)
    {
        ets_printf("--------------START------------------\n");
        free_heap_size = esp_get_free_heap_size();
        ets_printf("free_heap_size %d ",free_heap_size);

        start_ble_func();
        free_heap_size = esp_get_free_heap_size();
        ets_printf("after start_ble_func %d ",free_heap_size);
        vTaskDelay(100);
        stop_ble_func();
        free_heap_size = esp_get_free_heap_size();
        ets_printf("after stop_ble_func %d \n",free_heap_size);
        ets_printf("--------------END----------------\n\n");
    }
    return;
}
```

Alvin1Zhang, 2020-01-15:
@edubadia Thanks for reporting. Did you still have the issue? Thanks.

jack0c, 2023-04-12:
Will close this since long time no update and we have https://github.com/espressif/esp-idf/issues/7653