espressif/esp-idf #4166: esotic UART baudrates won't work

aster94, 2019-10-06:
## Environment

- Development Kit:     lolin32
- IDF version  lasted used in the arduino framework
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

If i set a custom baudrate in the UART it won't work as expected. At first we tought the problem was in the arduino core (https://github.com/espressif/arduino-esp32/issues/2004), but maybe it is not so I have been suggested to open an iussue here

### Code to reproduce this issue

```cpp
#include <Arduino.h>

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/uart.h"

#define ECHO_TEST_TXD (UART_PIN_NO_CHANGE)
#define ECHO_TEST_RXD (UART_PIN_NO_CHANGE)
#define ECHO_TEST_RTS (UART_PIN_NO_CHANGE)
#define ECHO_TEST_CTS (UART_PIN_NO_CHANGE)

#define BUF_SIZE 1024

void setup()
{
  uart_config_t uart_config = {
      .baud_rate = 10400,
      .data_bits = UART_DATA_8_BITS,
      .parity = UART_PARITY_DISABLE,
      .stop_bits = UART_STOP_BITS_1,
      .flow_ctrl = UART_HW_FLOWCTRL_DISABLE};
  uart_param_config(UART_NUM_2, &uart_config);
  uart_set_pin(UART_NUM_2, ECHO_TEST_TXD, ECHO_TEST_RXD, ECHO_TEST_RTS, ECHO_TEST_CTS);
  uart_driver_install(UART_NUM_2, BUF_SIZE * 2, 0, 0, NULL, 0);
}

void loop()
{
  const char *data = "test";
  uart_write_bytes(UART_NUM_2, data, 5);
  delay(2);
}
```

And here there are two logic analyzer data, the `esp-idf.sr` comes from the code above
[test.zip](https://github.com/espressif/esp-idf/files/3694140/test.zip)

koobest, 2019-10-08:
Hi, @aster94 
The clock divider register is 20-bit width(1048576), If  you use APB clock(80M), we can not get the baud-rate of 10400, but we can use REF_TICK(1M) as the source clock to get such a low baud-rate. When using IDF code, you did not specify the parameter `use_ref_tick`, so the default value is true. But arduino uart lib does not seem to support `ref_tick`, the source clock will always be APB_CLK.

thanks !!

negativekelvin, 2019-10-08:
@koobest  80M/0xFFFFF = 76

negativekelvin, 2019-10-08:
It takes 5ms to send 5 bytes at 10400 so if you delay only 2ms you are going to overflow the tx buffer.  But other than that, 10400 seems to work fine on uart 1&2.

aster94, 2020-04-06:
on the arduino core it was solved by https://github.com/espressif/arduino-esp32/pull/3664
I am closing since i guess that also here have been solved but feel free to reopen it

closed 2020-04-06 by aster94