espressif/esp-idf #4168: v4.0-beta1 'idf_as_lib' example fails to link for the Host target

WayneKeenan, 2019-10-06:
I'm on macOS and have tried to compile the OOTB example using clang and gcc but always get the error below. the 'build-esp32.sh' works ok btw.

```[ 87%] Building C object CMakeFiles/idf_as_lib.elf.dir/main.c.o
[100%] Linking C executable idf_as_lib.elf
Undefined symbols for architecture x86_64:
  "app_main", referenced from:
     -u command line option
     (maybe you meant: _app_main)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
gmake[2]: *** [CMakeFiles/idf_as_lib.elf.dir/build.make:86: idf_as_lib.elf] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:74: CMakeFiles/idf_as_lib.elf.dir/all] Error 2
gmake: *** [Makefile:84: all] Error 2
```

renzbagaporo, 2019-10-08:
Hi, just a clarification: You mention that gcc does not work as well, what does the error look like for it?

WayneKeenan, 2019-10-10:
```
./build.sh 
-- The C compiler identification is GNU 8.3.0
-- Checking whether C compiler has -isysroot
-- Checking whether C compiler has -isysroot - yes
-- Checking whether C compiler supports OSX deployment target flag
-- Checking whether C compiler supports OSX deployment target flag - yes
-- Check for working C compiler: /usr/local/bin/gcc-8
-- Check for working C compiler: /usr/local/bin/gcc-8 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/wayne/local.projects/NanoCast/src/Nanocast_Platform/toolchain/esp/esp-idf/examples/build_system/cmake/idf_as_lib.test/build
Scanning dependencies of target stub_freertos
[ 12%] Building C object stubs/freertos/CMakeFiles/stub_freertos.dir/task.c.o
[ 25%] Linking C static library libstub_freertos.a
[ 25%] Built target stub_freertos
Scanning dependencies of target stub_esp32
[ 37%] Building C object stubs/esp32/CMakeFiles/stub_esp32.dir/system_api.c.o
[ 50%] Building C object stubs/esp32/CMakeFiles/stub_esp32.dir/flash_ops.c.o
[ 62%] Building C object stubs/esp32/CMakeFiles/stub_esp32.dir/cpu_start.c.o
[ 75%] Linking C static library libstub_esp32.a
[ 75%] Built target stub_esp32
Scanning dependencies of target idf_as_lib.elf
[ 87%] Building C object CMakeFiles/idf_as_lib.elf.dir/main.c.o
[100%] Linking C executable idf_as_lib.elf
Undefined symbols for architecture x86_64:
  "app_main", referenced from:
     -u command line option
     (maybe you meant: _app_main)
ld: symbol(s) not found for architecture x86_64
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/idf_as_lib.elf.dir/build.make:86: idf_as_lib.elf] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:74: CMakeFiles/idf_as_lib.elf.dir/all] Error 2
gmake: *** [Makefile:84: all] Error 2
```

WayneKeenan, 2019-10-10:
The linker being used in both cases is the same macOS native one: https://github.com/Homebrew/homebrew-core/issues/17794#issuecomment-328174395

renzbagaporo, 2019-11-03:
Can you do an objdump on `main.c.o`?. It should be in the $BUILD_DIR/CMakeFiles/idf_as_lib.elf.dir:

```
objdump -t main.c.o
```

WayneKeenan, 2019-11-06:
Built with gcc-8:

```
build/CMakeFiles/idf_as_lib.elf.dir/main.c.o:	file format Mach-O 64-bit x86-64

SYMBOL TABLE:
0000000000000140 l       __TEXT,__cstring	lC0
000000000000014d l       __TEXT,__cstring	lC1
0000000000000152 l       __TEXT,__cstring	lC2
0000000000000153 l       __TEXT,__cstring	lC3
0000000000000158 l       __TEXT,__cstring	lC4
0000000000000189 l       __TEXT,__cstring	lC5
000000000000019f l       __TEXT,__cstring	lC6
00000000000001a8 l       __TEXT,__cstring	lC7
00000000000001b1 l       __TEXT,__cstring	lC8
00000000000001c0 l       __TEXT,__cstring	lC9
00000000000001dd l       __TEXT,__cstring	lC10
00000000000001f0 l       __TEXT,__eh_frame	EH_frame1
0000000000000000 g     F __TEXT,__text	_app_main
0000000000000000         *UND*	___stdoutp
0000000000000000         *UND*	_esp_chip_info
0000000000000000         *UND*	_esp_restart
0000000000000000         *UND*	_fflush
0000000000000000         *UND*	_printf
0000000000000000         *UND*	_puts
0000000000000000         *UND*	_spi_flash_get_chip_size
0000000000000000         *UND*	_vTaskDelay
```

Built with clang:

```

build/CMakeFiles/idf_as_lib.elf.dir/main.c.o:	file format Mach-O 64-bit x86-64

SYMBOL TABLE:
0000000000000000 g     F __TEXT,__text	_app_main
0000000000000000         *UND*	___stdoutp
0000000000000000         *UND*	_esp_chip_info
0000000000000000         *UND*	_esp_restart
0000000000000000         *UND*	_fflush
0000000000000000         *UND*	_printf
0000000000000000         *UND*	_spi_flash_get_chip_size
0000000000000000         *UND*	_vTaskDelay
```

renzbagaporo, 2019-11-08:
Can you try modifying `app_main` to `_app_main` in https://github.com/espressif/esp-idf/blob/a45e9985344575a80acb1dc9c12e7bec4d8af401/examples/build_system/cmake/idf_as_lib/stubs/esp32/CMakeLists.txt#L5? @WayneKeenan

renzbagaporo, 2019-11-11:
Hi. A fix for this is underway. In the meantime, the previous suggestion can serve as a workaround @WayneKeenan

WayneKeenan, 2019-11-11:
Thanks, that worked.

closed 2019-11-19 by espressif-bot