espressif/esp-idf #4169: sending frame with a while time and system get crash on websokcet

huazhenjiang, 2019-10-07:
### Environment
Development Kit: ESP32-DevKitC
Kit version (for WroverKit/PicoKit/DevKitC): v2
Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it): v4.0-released
Operating System: Windows 10
Power Supply: USB

###Problem Description
I try the websoket example to connect wss server and send a binary data about 50 bytes per 50ms. Also I modify the mask bit for each sending frame. Then the system get crash and auto reboot about 10 minutes later. The following log show it get a 'esp-tls:write error' first then get a 'esp-tls: read error', disconnect finally. It seems some memory address can not be writable but it occured.

###Expected Behavior
sending packets continuously
###Actual Behavior
system crash and auto reboot
###Steps to repropduce

###Code to reproduce this issue
none.

###Debug Logs
E (775041) esp-tls: write error :-27648:
E (775041) TRANS_SSL: esp_tls_conn_write error, errno=Success

E (775661) esp-tls: read error :-30592:
E (775661) TRANS_SSL: esp_tls_conn_read error, errno=Success
E (775661) TRANSPORT_WS: Error read data
E (775671) WEBSOCKET_CLIENT: Error read data
E (775671) esp-tls: write error :-27648:
I (775671) WEBSOCKET_CLIENT: Reconnect after 10000 ms
E (775681) TRANS_SSL: esp_tls_conn_write error, errno=Success
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4000c344  PS      : 0x00060730  A0      : 0x801186e1  A1      : 0x3ffcde00  
A2      : 0x3ffcbde4  A3      : 0xfefefefc  A4      : 0x0000000c  A5      : 0x3ffcbde4  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80000000  A9      : 0x3ffcdaa0  
A10     : 0x00000049  A11     : 0x00000002  A12     : 0x00000001  A13     : 0x3ffcdcb0  
A14     : 0x00000049  A15     : 0x3ffcdcb0  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfefefefc  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

ELF file SHA256: 95686d089b74a9f15266d91d78e6e37e0539990f031642a2da9aa14dbe45ae1b

Backtrace: 0x4000c341:0x3ffcde00 |<-CORRUPTED

Rebooting...
ets Jun  8 2016 00:22:57

david-cermak, 2019-10-07:
Hi @huazhenjiang 

Thanks for reporting this issue! Seems to be a race condition as the client first checks [if connected](https://github.com/espressif/esp-idf/blob/master/components/esp_websocket_client/esp_websocket_client.c#L542) and locks [afterwards](https://github.com/espressif/esp-idf/blob/master/components/esp_websocket_client/esp_websocket_client.c#L552). You can test if the issue goes away moving locking before connection check.
Will fix this issue!

huazhenjiang, 2019-10-08:
Hi @david-cermak,

Do you mean moving the semaphoreTake like the following graph?
[image]

david-cermak, 2019-10-11:
@huazhenjiang Thanks for reporting this bug and sorry for not getting back to you sooner.
The fix seems a bit more complex that what was suggested here above. It will be delivered to the IDF as soon as formally approved and tested.

Meanwhile, posting a patch agains IDF-4.0 that should address these issues.
[0001-websocket_client-fix-locking-mechanism-in-ws-client-.patch.txt](https://github.com/espressif/esp-idf/files/3717877/0001-websocket_client-fix-locking-mechanism-in-ws-client-.patch.txt)

huazhenjiang, 2019-10-14:
@david-cermak Actually I would receive data every 50ms from another task(uart) which passing by message queue in my project. And then I patched but it seemed to be lock into the websocket task .
I could not see any data receive from uart task... is that expected?

huazhenjiang, 2019-10-17:
@david-cermak thank you, this patch looks like worked.

Alvin1Zhang, 2019-10-25:
@huazhenjiang Thanks for the testing and confirmation. Will reopen the issue and wait until our fix is merged, thanks.

closed 2019-11-14 by espressif-bot