espressif/esp-idf #4170: Modbus master

gbortnikov, 2019-10-07:
I receive device slave data only from esp32. I can not get data from other devices connected to the Modbus network, the request passes in the response only zero

Alvin1Zhang, 2019-10-08:
@gbortnikov Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

gbortnikov, 2019-10-08:
ESP32 is a master. I connect it to the slave device via the max485 interface. The request is sent but in response to the required information comes zero. I tried to test the master through the converter to the PC (where the PC acted as a slave), but also zero in response. But when I connect to another esp 32 everything works fine. Slave and master taken from standard examples of esp-idf
[image]
[image]
[image]

alisitsyn, 2019-10-08:
Hi gbortnikov,

According to your pictures the changes for Modbus master example are incorrect and you made several errors when tried to configure your master application. The Modbus_master example has been designed to work with certain ```device_parameters[]``` table. The trigger_operation_task() is the example task in sense_main.c file which performs the actual control process and displays the values of float format received from slaves. You changed the "Temperature_1" to type PARAM_TYPE_U8 but left size of parameter instance = 4 in master parameter table. So, the Modbus driver updates just one byte in the instance for the "Temperature_1" parameter and then task trying to print the actual float value from the 4 bytes of instance. In your host application you are writing random values as separate registers and this value does not correspond to valid float value.
Other "Temperature_1" parameters with different size stores its values into the same instance of "mb_device1_temperature" overriding the value of previously read parameters.
I propose you to perform code changes as below:
1. Make sure the CID for each parameter is unique.
2. ```sense_main.c: line 107:
                        // Update read value
                        ESP_LOGI(SENSE_MAIN_TAG, "cid: %u, %s(%s) = 0x%x", (uint16_t)cid,
                                                (char*)cid_data.param_key,
                                                (char*)cid_data.param_units,
                                                *(uint32_t*)&value);
``` 
This will allow to display parameter instance as 4 byte hex value which will correspond to values in your host program. It is possible to display the value according to its type in the master parameter table but it is responsibility of user to handle this adopted code according to actual process.
3. Make sure also that each parameter defined in the "Instance Offset" of ```device_parameters[]``` has correct size in the structure and "Data Size" corresponds to its size. You can also use zero for instance offset then the instance will be created automatically for the parameter according to its size.

--
Alex

closed 2020-08-19 by Alvin1Zhang