espressif/esp-idf #4172: Port of net_sockets in mbedtls component does not have poll function

crippsJack, 2019-10-07:
The ESP-IDF port of `net_sockets` does not have a port of the `mbedtls_net_poll()` function. This is a requirement of the AWS-SDK v4 mbedtls wrapper, so is there any plan to add this function or is it not possible due to constraints of the ESP32 platform, in which case do you have a recommended solution?

mahavirj, 2019-10-08:
@crippsJack If you copy/use `mbedtls_net_poll` from https://github.com/espressif/mbedtls/blob/f5f2e5926cd294ae7cb579ff6a12ad9303caeb6e/library/net_sockets.c#L456, does it help to unblock your use-case?

crippsJack, 2019-10-09:
I will give this a try, thanks

mahavirj, 2020-06-22:
Closing due to lack of response. Please feel to re-open if any further help is required,

closed 2020-06-22 by mahavirj

dragomat, 2024-02-20:
Is there any reason why `mbedtls_net_poll()` has never been included in ESP-IDF? 
I found myself in the same situation as described above (stuck waiting in ssl_write with no possibility of enabling the timeout) and even in the newest IDF versions (5.1 as well) it seems that we need to implement our own version of that function. 

Thanks