espressif/esp-idf #4173: implementation status of Clock Stretching on ESP32

almoondsllc, 2019-10-07:
Hello,
I need to use the iAQCore chip with I2C that require Clock Stretching https://www.i2c-bus.org/clock-stretching/(https://ams.com/documents/20143/36005/iAQ-core_DS000334_1-00.pdf).

It seems that while Arduino implements this with something as easy as   Wire.setClockStretchLimit(1000) the esp-idf has nothing similar.

So will/is clock stretching feautures implemented in the actual esp-idf code/library? Thanks.

Alessandro

negativekelvin, 2019-10-07:
Supported up to 13ms

https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/i2c.html#_CPPv415i2c_set_timeout10i2c_port_ti

stickbreaker, 2019-10-08:
@almoondsllc the only way to extend the clock stretching hardware limit of 13ms is to reduce the processor clock speed. the maximum timeout period is 2**20 APB clock cycles.  APB clock is either 80MHz or CPU clock, whichever is slower. But, WiFi/Bluetooth does not function if APB clock is less than 80MHz.

costaud, 2020-05-26:
The timeout for i2c master is similar to clock stretching, but different for i2c slave side.
Seems esp32 slave doesn't support hardware clock stretching. (which will be supported on esp32s2 and later chips)

closed 2020-05-26 by costaud