espressif/esp-idf #4174: new api request for http client

w19787, 2019-10-08:
in order to use one http client for multiple http performs, we should have an api to clean up http header settings.

currently, there are only follow methods for header setting:
esp_err_t esp_http_client_delete_header(esp_http_client_handle_t client, const char *key);
esp_err_t esp_http_client_get_header(esp_http_client_handle_t client, const char *key, char **value);
esp_err_t esp_http_client_delete_header(esp_http_client_handle_t client, const char *key);

i would suggest to add an additional api for header:
esp_err_t esp_http_client_cleanup_header(esp_http_client_handle_t client);

thanks

shubhamkulkarni97, 2020-01-02:
@w19787 Can you share your test-code or use-case which would fail if the suggested api is not implemented?

Alvin1Zhang, 2020-01-15:
@w19787 Thanks for reporting, would you please help if any updates for this issue? Thanks.

Alvin1Zhang, 2020-03-05:
@w19787 Thanks for reporting. Will close the ticket due to short of feedback. Feel free to reopen with more details. Thanks.

closed 2020-03-05 by Alvin1Zhang