espressif/esp-idf #4175: cc1plus: some warnings being treated as errors

oib, 2019-10-08:
Debian Testing/Bullseye
https://gitlab.com/teamoskar/oskar_esp32.git

oib@debian:~/esp/oskar_esp32$ make flash
Toolchain path: /home/oib/.espressif/tools/xtensa-esp32-elf/esp32-
2019r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
Python requirements from /home/oib/esp/esp-idf/requirements.txt are
satisfied.

App "hidd_demos" version: 8283414
CXX build/nkolban_BLE/GeneralUtils.o
In file included from
/home/oib/esp/oskar_esp32/components/nkolban_BLE/GeneralUtils.cpp:16:
/home/oib/esp/oskar_esp32/components/nkolban_BLE/FreeRTOS.h:60:28:
error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type =
RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
/home/oib/esp/oskar_esp32/components/nkolban_BLE/GeneralUtils.cpp: In
static member function 'static const char*
GeneralUtils::wifiErrorToString(uint8_t)':
/home/oib/esp/oskar_esp32/components/nkolban_BLE/GeneralUtils.cpp:450:8
: error: enumeration value 'WIFI_REASON_CONNECTION_FAIL' not handled in
switch [-Werror=switch]
  switch((wifi_err_reason_t) errCode) {
        ^
cc1plus: some warnings being treated as errors
make[1]: *** [/home/oib/esp/esp-idf/make/component_wrapper.mk:292:
GeneralUtils.o] Error 1
make: *** [/home/oib/esp/esp-idf/make/project.mk:604: component-
nkolban_BLE-build] Error 2

Alvin1Zhang, 2019-10-08:
@oib Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

negativekelvin, 2019-10-08:
If you're going to build a third party codebase, you will either have to build against a supported version of esp-idf or update the codebase to be compatible with a newer esp-idf.  Ask for help on esp32.com.

benjaminaigner, 2019-10-08:
@oib This issue happened with my port of Neil Kolban's code as well (Oskar is using my project for HOG):
https://github.com/asterics/esp32_mouse_keyboard/issues/16

Possible solutions are there...

mws-rmain, 2019-10-15:
For a resolution of "error: 'ringbuf_type_t' has not been declared", see [esp32-snippets issue #889](https://github.com/nkolban/esp32-snippets/issues/889)

Alvin1Zhang, 2020-03-12:
@oib Thanks for reporting, for third party issues, you could raise a post on our forum at https://esp32.com, thanks.

closed 2020-03-12 by Alvin1Zhang