espressif/esp-idf #4176: How to use iconv on ESP-IDF

JaeHoLee1811, 2019-10-08:
Hello!
How to use iconv on esp-idf or esp32?

I tried to use iconv but I get this error  help me!! please 
I used 4days to this problem  :(

    int ret;
    iconv_t it;
    char ksc_buf[1024];
	strcpy(ksc_buf, data);
     
    // KSC(완성형) 코드를 UTF(유니코드)로 변환하면 원래 크기보다 커지므로 크게~
    char utf_buf[1024] = {0x00, };
    size_t in_size, out_size;

    memset(utf_buf, '\0', 1024);
 
    // 어떤 시스템에서는 char** 가 아니라 const char** 인 경우도 있음
    char *input_buf_ptr = ksc_buf;
    char *output_buf_ptr = utf_buf;
 
    in_size = strlen(ksc_buf);
    out_size = sizeof(utf_buf);
 
    it = iconv_open("UTF-8", "EUC-KR"); // EUC-KR을 UTF-8로
    ret = iconv(it, &input_buf_ptr, &in_size, &output_buf_ptr, &out_size);
     
    if (ret < 0)
    {
        printf("ret : %d, errno : %d\n", ret, errno);
        return(-1);
    }
    else
    {
        printf("[%s](%d) => [%s][(%d)\n",
            ksc_buf, in_size, utf_buf, out_size);
    }
     
    iconv_close(it);

c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/msys32/home/esp/blufi/build/main\libmain.a(blufi_example_main.o):(.literal.Change+0x18): undefined reference to `iconv_open'
c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/msys32/home/esp/blufi/build/main\libmain.a(blufi_example_main.o):(.literal.Change+0x1c): undefined reference to `iconv'
c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/msys32/home/esp/blufi/build/main\libmain.a(blufi_example_main.o):(.literal.Change+0x20): undefined reference to `iconv_close'
c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/msys32/home/esp/blufi/build/main\libmain.a(blufi_example_main.o): in function `Change':
C:/msys32/home/esp/blufi/main/blufi_example_main.c:534: undefined reference to `iconv_open'
c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/msys32/home/esp/blufi/main/blufi_example_main.c:535: undefined reference to `iconv'
c:/msys32/opt/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/msys32/home/esp/blufi/main/blufi_example_main.c:544: undefined reference to `iconv_close'
collect2.exe: error: ld returned 1 exit status
make: *** [C:/msys32/home/esp/esp-idf/make/project.mk:523: /home/esp/blufi/build/blufi_demo.elf] error

Alvin1Zhang, 2019-10-08:
@JaeHoLee1811 Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. **Also we suggest that you could raise a post on forum https://esp32.com for "how" questions**. Thanks.

lbernstone, 2020-11-25:
This is still an issue.  I am using the arduino-esp32 idf-release/v4.2 branch, which is based on idf commit 494a124d9 (pretty close to tip of master, using 2020r3 toolchain).  @JaeHoLee1811's output log and script are close enough to what I am trying to do (get glib working).  This looks to me like the toolchain does not include the functions for the exfun'd symbols in newlib/include/iconv.h.  Happens with both esp32 and esp32s2.

VojtechBartoska, 2022-04-27:
Hello,

sorry for answering late but we have missed this one. Is it still valid?

ESP-Marius, 2024-11-15:
`iconv` seems to be available in the toolchain now and the example in this issue can build successfully.

closed 2024-11-15 by ESP-Marius