espressif/esp-idf #4177: i2c doesn't work when spi enabled - RA8875 and FT5206GE1

hfmartinez, 2019-10-08:
good afternoon, at this moment an application is developed with a touch panel with the RA8875 driver, we currently communicate through the ESP-IDF with the SPI protocol, additionally we have a capacitive touch panel with FT5206GE1 driver through the I2C protocol, when we test The separate drivers work without any problem, but when they are created in the same project in different tasks, problems are generated with the communication with each driver, I would appreciate any information you can give me. Thanks a lot.

Alvin1Zhang, 2019-10-09:
@hfmartinez Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

mehrdad987, 2019-11-06:
same problem but maybe its because of my ISR interrupt
Module: ESP32 DEV --with SD (LVC1225A)Module and RTC(DS3231) 
the program reading the sensors data in FREERTOSTASK  that call with ISR 
in one task I making string off all data and RTC time then  append them to the sd card
after a minute I encounter with guru meditation(load prohibited) and also sometimes my time is going to wrong number sam as 2019:10:45 30:45

RTC:
`
void taskOne( void * parameter )
{
  RtcDateTime now = Rtc.GetDateTime();
  printDateTime(now);
  RtcTemperature temp = Rtc.GetTemperature();
  //float Device_Temperature = 
  //temp.Print(Serial);
  // you may also get the temperature as a float and print it
  Device_Temperature = (temp.AsFloatDegC());
  vTaskDelete( NULL );
}`

`  
void taskTwo( void * parameter )
{
forsdstr = SN + ',' + timeSec + ',' + ex_pressure + ',' + TEMP1 + ',' + TEMP2 + ',' + string_rpm + ','   + O2 + ',' + LAM + ','  + Latitude + ',' + Longitude + ',' + Speed + "\n";
  forsd = forsdstr.c_str();
  
  appendFile(SD, today_file, forsd);  
  Serial.println("-------------------APPEND-TSK 1");
  vTaskDelete( NULL );
}`

Alvin1Zhang, 2020-01-15:
@hfmartinez Thanks for reporting, would you please help share if any updates for the issue? Thanks.

costaud, 2020-05-26:
We failed to reproduce this issue with given information, close this issue for now. 
Feel free to reopen it if there is more further information.

closed 2020-05-26 by costaud