espressif/esp-idf #4180: GPIO GPIO_INTR_ANYEDGE ISR gets triggered multiple time with the same state

lohisynths, 2019-10-09:
## Environment

- Development Kit:      [DOIT Esp32 DevKit]
- Kit version           [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version           [3.3]
- Build System:         [Make]
- Compiler version      [crosstool-ng-1.22.0-80-g6c4433a]
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
While using interrupts on both edges, ISR gets triggered with the same GPIO state one after another.
Probably there has to be some way for distinguishing falling / rising edge of GPIO_INTR_ANYEDGE ISR.

### Expected Behavior
ISR should gets triggered only if new GPIO state occurs.
There should be no situation, where the same state ISR gets triggered multiple times one after another.

### Actual Behavior
ISR gets called probably with the same state of GPIO more than once, or one of GPIO ISR gets lost.

### Steps to repropduce

1. Compile program with esp-idf v3.3
2. Click on Boot button on DOIT Esp32 DevKit devkit
3. After multiple clicks, you will get an error because of actual state of gpio is the same as previous one.

### Code to reproduce this issue

Esp-idf code
```cpp
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"

#define GPIO_INPUT_IO    (gpio_num_t)(0)
#define GPIO_INPUT_PIN_SEL  ((1ULL<<GPIO_INPUT_IO) )
#define ESP_INTR_FLAG_DEFAULT ESP_INTR_FLAG_LEVEL1

static void IRAM_ATTR gpio_isr_handler(void *arg) {
  static bool last_state = 1;
  gpio_num_t gpio_num = (gpio_num_t) ((int) arg);
  bool state = gpio_get_level(gpio_num);

  if (last_state == state) {
    ets_printf("error, last_state == state = %d\r\n", state);
  }
  last_state = state;
}

extern "C" void app_main() {
  gpio_config_t io_conf = { };
  io_conf.intr_type = GPIO_INTR_ANYEDGE;
  io_conf.pin_bit_mask = GPIO_INPUT_PIN_SEL;
  io_conf.mode = GPIO_MODE_INPUT;
  io_conf.pull_up_en = GPIO_PULLUP_ENABLE;

  esp_err_t ret = gpio_config(&io_conf);
  if(ret != ESP_OK) {
    printf("error, gpio_config failed\r\n");
  }

  ret = gpio_install_isr_service(ESP_INTR_FLAG_IRAM);
  if(ret != ESP_OK) {
    printf("error, gpio_install_isr_service failed\r\n");
  }

  ret = gpio_isr_handler_add(GPIO_INPUT_IO, gpio_isr_handler, (void*)GPIO_INPUT_IO);
  if(ret != ESP_OK) {
    printf("error, gpio_isr_handler_add failed\r\n");
  }

  for (;;) {
    vTaskDelay(pdMS_TO_TICKS(200));
  }
}
```

Arduino code:
```cpp
const byte interruptPin = 0;
volatile byte last_state = HIGH;

void setup() {
  Serial.begin(115200);
  pinMode(interruptPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin), isr, CHANGE);
}

void loop() {
  delay(200);
}

void isr() {
  byte state = digitalRead(interruptPin);
  if(state == last_state) {
    Serial.println("error!");
  }
  last_state = state;
}
```

negativekelvin, 2019-10-10:
You'll either have to add some hardware hysteresis in front of the gpio or write a smarter interrupt handler to filter out false edges.

lohisynths, 2019-10-10:
Sorry, but if there was a problem with hysteresis, incoming states of GPIO should be toggled every time the new ISR gets called.

lohisynths, 2019-10-10:
So, can you provide one? By hardware you mean PCNT + pcnt_filter_enable()? This is a quite stable option, but there are only 8 channels available from what I know.
External decoupling caps + resistors don't work in our case, as it is highly depended on the slope rate of rising/falling edge, so in a worst-case scenario, ISR will get triggered multiple times on voltage threshold level - because of no hysteresis inside the chip.
When it comes to software implementation, what can you propose? Constantly sampling GPIO pin from other thread, ignoring switch bounces that occur before 'stabilization'. We can, of course, implement this behavior inside ISR, but adding another button will scale quickly to a large amount of ISR occurring just to check if the last change of state occurs longer than 'debounce time'. Which is also not ideal. We think about adding extra Schmidt trigger, at every input that could 'bounce', but it will introduce some changes to the hardware, which we like to avoid.

filzek, 2019-10-15:
Not quite, you need to know state, and also have a time to change, in your routine

lohisynths, 2019-10-15:
What do you mean? Can you elaborate?

costaud, 2019-10-28:
If you are going to realize a simple button by GPIO_INTR_ANYEDGE, maybe you could refer to this example https://github.com/espressif/esp-iot-solution/tree/master/components/general/button

koobest, 2019-11-04:
@lohisynths 
Have you solved this issue?

Alvin1Zhang, 2020-01-15:
@lohisynths Thanks for reporting. Would you please help share if any updates for the issue? Has your issue been resolved? Thanks.

Alvin1Zhang, 2020-05-28:
@lohisynths Thanks for reporting, hope details provided by @costaud could help in resolving the issue. Will close due to short of feedback, feel free to reopen with more details or updates. Thanks.

closed 2020-05-28 by Alvin1Zhang

negativekelvin, 2020-05-28:
There is a new errata 3.14 about this

shivdeepak, 2022-08-13:
FWIW, the following code works with a 50Khz square wave. Look at errata [3.14](https://espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf) as @negativekelvin mentioned, if the code doesn't work for you.

```c
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"

#define GPIO_OUTPUT          10
#define GPIO_INPUT            4

static void IRAM_ATTR gpio_isr_handler(void *arg) {
    gpio_set_level(GPIO_OUTPUT, gpio_get_level((gpio_num_t) ((int) arg)));
}

void app_main(void) {
    ESP_ERROR_CHECK_WITHOUT_ABORT(gpio_set_direction(GPIO_OUTPUT, GPIO_MODE_OUTPUT));
    ESP_ERROR_CHECK_WITHOUT_ABORT(gpio_set_direction(GPIO_INPUT, GPIO_MODE_INPUT));
    ESP_ERROR_CHECK_WITHOUT_ABORT(gpio_set_intr_type(GPIO_INPUT, GPIO_INTR_ANYEDGE));
    ESP_ERROR_CHECK_WITHOUT_ABORT(gpio_install_isr_service(ESP_INTR_FLAG_IRAM));
    ESP_ERROR_CHECK_WITHOUT_ABORT(gpio_isr_handler_add(GPIO_INPUT, gpio_isr_handler, (void*)GPIO_INPUT));

    for (;;) {
        vTaskDelay(pdMS_TO_TICKS(200));
    }
}
```