espressif/esp-idf #4181: GPIO reset to initial state

tatulea, 2019-10-09:
Hi,

I have a project with a Type-C connector to a custom board. This means that my communication pins need to be configured based on the Type-C insertion way (face up or face down). I am doing some debugging because my Type-C connection doesn't work all the time and I would like to know how to get the initial state of some GPIO pins (input or output; pull-up state, etc). I am using gpio_reset_pin to delete the configuration of a GPIO, but I am not sure if this function sets its really initial state.

I posted this question on forum as well, but I have no response (https://www.esp32.com/viewtopic.php?f=2&t=12609)

negativekelvin, 2019-10-09:
Initial state depends on the gpio, just always set the state you want.  You can see exactly what gpio_reset_pin does

https://github.com/espressif/esp-idf/blob/879778a6772dfcb5f78ac458ae2a2a58e34e400e/components/driver/gpio.c#L333-L346

tatulea, 2019-10-09:
Can I read the config for a GPIO? Or is there any info about the initial state of each pin?

negativekelvin, 2019-10-10:
In the datasheet section a.4.  I don't think there is api to get gpio config so you would have to read registers.

closed 2019-10-15 by tatulea