espressif/esp-idf #4185: I2S available method

howroyd, 2019-10-10:
Quick suggestion:

It would be useful to have an `i2s.available()` method/function.  At the moment, the only way to see if there are bytes to be read is to call `read()`, which reports a misleading `ESP_ERR_INVALID_ARG` error if the rx buffer is empty.

Alvin1Zhang, 2019-10-11:
@howroyd Thanks for the suggestion.

miketeachman, 2020-02-11:
I could use this as well.  I implemented an `i2s_read_peek()` function in my ESP-IDF fork to determine the number of available bytes.
https://github.com/miketeachman/esp-idf/commit/f4843a6001e98bfc246c478b010bb247682b60e7

```
esp_err_t i2s_read_peek(i2s_port_t i2s_num, size_t *bytes_available)
{
    *bytes_available = 0;
    I2S_CHECK((i2s_num < I2S_NUM_MAX), "i2s_num error", ESP_ERR_INVALID_ARG);
    I2S_CHECK((p_i2s_obj[i2s_num]->rx), "rx NULL", ESP_ERR_INVALID_ARG);
    xSemaphoreTake(p_i2s_obj[i2s_num]->rx->mux, (portTickType)portMAX_DELAY);

    int residual_dma_bytes;
    if (p_i2s_obj[i2s_num]->rx->curr_ptr == NULL) {
        residual_dma_bytes = 0;
    } else {
        residual_dma_bytes = p_i2s_obj[i2s_num]->rx->buf_size - p_i2s_obj[i2s_num]->rx->rw_pos;
    }

    int num_full_dma_bufs = uxQueueMessagesWaiting(p_i2s_obj[i2s_num]->rx->queue);

    ESP_LOGD(I2S_TAG, "uxQueueMessagesWaiting: %d", num_full_dma_bufs);

    *bytes_available = (num_full_dma_bufs * p_i2s_obj[i2s_num]->rx->buf_size) + residual_dma_bytes;
    xSemaphoreGive(p_i2s_obj[i2s_num]->rx->mux);
    return ESP_OK;
}
```