espressif/esp-idf #4187: UART events

besso-exceline, 2019-10-11:
Hi,

cant get working pattern event, 
example uart_events_example_main.c

only pattern gets detected, data part is always empty;
Sending 8 charcters to ESP32 with leading +++

ESP_LOGI(TAG, "read pat : %s", pat); prints the received pattern +++ correctly
however 
ESP_LOGI(TAG, "read data: %s", dtmp); dos not print anything,

Alvin1Zhang, 2019-10-12:
@besso-exceline Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

negativekelvin, 2019-10-12:
Unable to reproduce.  Did you change EX_UART_NUM?  If so the data is echoed on that port not the console port.

besso-exceline, 2019-10-12:
Hi, thanks for reply. Using ESP32 and IDF version 3.3

I understand that patternt is detected at UART_PATTERN_DET event, and UART_DATA event captures data part; But what is the point of having   ESP_LOGI(TAG, "read data: %s", dtmp) in UART_PATTERN_DET event? It does not print data part; Data part is printed in UART_DATA event

negativekelvin, 2019-10-12:
If the data comes before the pattern there will be data there, but yes it should check the length before printing.

besso-exceline, 2019-10-12:
When I send following message to the ESP32:
+++MyCommand
Then 'case UART_PATTERN_DET' prints +++ but not MyCommand
When I send multiple like 
+++MyCommand+++MyCommand+++MyCommand
Then it prints last 2 MyCommand, but first one is always missing

Using the example as it is in ESP-IDF reposotory without changes and cant achieve the desired result.
I am sending messages to ESP32 from Windows PC .net SerialPort class.

negativekelvin, 2019-10-12:
If I send `+++abcd+++efgh+++ijkl`

Then I get

data:
pat: +++

data: abcd
pat: +++

data: efgh
pat: +++

DATA EVT: ijkl

Pattern detect is more useful if the pattern comes at the end of a command like '\r'

besso-exceline, 2019-10-14:
OK, pattern need to be at end of command string then and not in the beginning; Would be nice to have a short hint in ReadMe about pattern detection, bcs that is the very interesting feature in UART of ESP32

closed 2019-11-14 by espressif-bot