espressif/esp-idf #4188: Blufi for NimBLE?

copercini, 2019-10-12:
Hi!

I'm quite impressed with Nimble's lightness, you can run the whole BLE core with less than 10k of heap and only 300Kb of flash while in bluedroid it's easily reaches 70k+ of heap and 650Kb+ of space

Well, I would like to port my whole project to nimble, but only Blufi if only available on bluedroid. Is there any plan to port Blufi for NimBLE too?

dhrishi, 2019-10-14:
@copercini  NimBLE is not ported for Blufi and there is no plan to do so. Instead, we have added its support in the Unified Provisioning framework (similar functionality). You can read the details about the architecture, process and security here: https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/provisioning/provisioning.html#unified-provisioning

You can try it out using the example: https://github.com/espressif/esp-idf/tree/master/examples/provisioning/ble_prov. Please refer README.md in the same directory for more details.

Note: You will have to select NimBLE host in the menuconfig `make menuconfig -> Component config -> Bluetooth -> Bluetooth Host -> NimBLE - BLE only` and then compile the example as currently we have kept Bluedroid as the default option.

copercini, 2019-10-14:
Thanks for the answer, Unified Provisioning framework seems nice, but unfortunately I'll have to port blufi to nimble, because we already have the android/iOS apps ready for blufi

closed 2019-10-14 by copercini

dhrishi, 2019-10-16:
@copercini Just to close on this, we have Android and iOS apps for unified provisioning as well. You can see the apk as well as pointers to playstore:

https://github.com/espressif/esp-idf-provisioning-android/releases/tag/release_v1.2

https://github.com/espressif/esp-idf-provisioning-ios/releases/tag/release_v1.2

copercini, 2019-10-16:
@dhrishi yes the apps are cool, but unfortunately in my case we already have our own custom app based on blufi, that already works for our other ESP32 based products. So I don't have many choices, or port blufi to NimBLE or change the main code to bluedroid. Right now I'm looking to only try port blufi simplified (from ESP32  Homekit SDK)

dhrishi, 2019-10-16:
@copercini  No worries :-) I added this so that it would be helpful to users browsing this issue :-D

venkateshgr-ayla, 2021-06-08:
Hi @copercini were you successful in porting Blufi to Nimble?
Even am on the same track to do so. Any help or suggestions?

dhrishi, 2021-06-08:
Hi @copercini @venkateshgr-ayla  There's already a PR to add support of Blufi using NimBLE host: https://github.com/espressif/esp-idf/pull/6904

It is under review right now. We will update when it gets merged.

venkateshgr-ayla, 2021-06-08:
Hi Hrishikesh,

That’s indeed a great news. Any rough estimate when it would be available for us? So that we can plan accordingly.

Thanks,
GR.Venkatesh

Reply to: espressif/esp-idf ***@***.***>
Date: Tuesday, 8 June 2021 at 11:58 AM
To: espressif/esp-idf ***@***.***>
Cc: Venkatesh GR ***@***.***>, Mention ***@***.***>

dhrishi, 2021-06-08:
@venkateshgr-ayla It would be available in ESP-IDF master branch in a few days. For now, you can just try it out from the PR branch (It should work fine)