espressif/esp-idf #4189: Use CMAKE_BUILD_TYPE instead of CONFIG_COMPILER_OPTIMIZATION_LEVEL_RELEASE

sysprogs, 2019-10-12:
Currently the ESP-IDF defines a special variables for controlling the optimization level (CONFIG_COMPILER_OPTIMIZATION_xxx). However, CMake already has a mechanism for distinguishing debug/release bulds (CMAKE_BUILD_TYPE) that could be set to Debug/Release/MinSizeRel.

It would be great if ESP-IDF could follow the CMAKE_BUILD_TYPE setting, instead of defining its own configuration variable for it, as many IDEs support independent debug/release builds out-of-the-box.

Note: you can programmatically check the current build type and adjust the options accordingly in case you want to tweak it beyond the default CMake logic:

```
if (CMAKE_BUILD_TYPE EQUAL "DEBUG")
[set optimization for debug build]
endif()
```

renzbagaporo, 2020-08-17:
Hi @sysprogs. Sorry for the super late follow up on this. `CONFIG_COMPILER_OPTIMIZATION_xxx` is a remnant of the transition to CMake from Make. However, we are looking into this and having discussions for the next batch of CMake improvements. This would be in my near-term things to implement.

zikalino, 2023-11-29:
Closing this ticket as it's already pretty old and we are not going to make this change.

closed 2023-11-29 by zikalino