espressif/esp-idf #4190: WiFi Driver , Wi-Fi Buffer Configure Not Work

wgcod, 2019-10-12:
Now, I use the sniffer mode of ESP32, but sometimes WiFi driver suddenly uses a lot of memory, causing device problems. I tried to configure the WIFI driver Buffer, as shown in the figure
[image]
But the problem still arises. I tested where the memory was used, as shown in the figure.
[image]
A large number of wifi_malloc, unlimited number of memory applications, and now I am sure that a large number of memory applications are here, until the application failed, using up all the memory in heap.
Now I want to limit some memory applications in WiFi dirver, it doesn't matter to lose packets. I tried some configurations in make menuconfig, but it didn't work. How should I operate now? Is the configuration incorrect?
I've measured that it might also have something to do with the AMPDU configuration, but I can't find the ideal configuration.
thank you.

Alvin1Zhang, 2019-10-12:
@wgcod Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

espxiehang, 2019-11-06:
@wgcod Can you give me a way to reproduce this issue?

Alvin1Zhang, 2020-01-15:
@wgcod Thanks for reporting, would you please help share if any updates of the issue? Thanks.

liuzfesp, 2020-02-21:
Hi @wgcod, could you help to provide following info:

1. What's the WiFi mode, STA, AP, STA+AP or NULL mode?
2. About how many memory has been allocated by WiFi when the problem happens?
3. Your implementation of the sniffer callback function, I want to know how the application handle the packets received by the sniffer callback.
4. How about configuring the Maximum WiFi Static RX Buffer number to 2, does the memory exhaust also?

Moreover,  configuring WiFi Dynamic RX Buffer Number will not impact sniffer, because sniffer directly allocate the memory from heap.

liuzfesp, 2020-04-30:
Hi @wgcod, any update about this issue?

Alvin1Zhang, 2020-05-28:
@wgcod Thanks for reporting and will close now due to short of feedback since reporting, feel free to reopen with more details. Thanks.

closed 2020-05-28 by Alvin1Zhang