espressif/esp-idf #4192: Request: Additional CMake API macros

WayneKeenan, 2019-10-12:
It would be handy if the IDF CMake API also had public macros for:

+ monitor
+ flash erase

I'm seeing ones for: app, app-flash, gen_project_binary, [project binary] et.c in the CLion IDE, it would great to be able to launch those too (especially flash)

copercini, 2019-10-13:
these options are already available:
```
C:\Users\copercini\Desktop\esp-idf>idf.py --help
Usage: idf.py [OPTIONS] COMMAND1 [ARGS]... [COMMAND2 [ARGS]...]...

  ESP-IDF build management

Options:
  --version                       Show IDF version and exit.
  -C, --project-dir PATH          Project directory.
  -B, --build-dir PATH            Build directory.
  -n, --no-warnings               Disable Cmake warnings.
  -v, --verbose                   Verbose build output.
  --ccache / --no-ccache          Use ccache in build. Disabled by default, unless IDF_CCACHE_ENABLE environment
                                  variable is set to a non-zero value.
  -G, --generator [Ninja|MinGW Makefiles]
                                  CMake generator.
  -D, --define-cache-entry TEXT   Create a cmake cache entry. This option can be used at most once either globally, or
                                  for one subcommand.
  -b, --baud INTEGER              Baud rate. This option can be used at most once either globally, or for one
                                  subcommand.
  -p, --port TEXT                 Serial port. This option can be used at most once either globally, or for one
                                  subcommand.
  --help                          Show this message and exit.

Commands:
  all                    Aliases: build. Build the project.
  app                    Build only the app.
  app-flash              Flash the app only.
  bootloader             Build only bootloader.
  bootloader-flash       Flash bootloader only.
  clean                  Delete build output files from the build directory.
  confserver             Run JSON configuration server.
  efuse_common_table     Genereate C-source for IDF's eFuse fields.
  efuse_custom_table     Genereate C-source for user's eFuse fields.
  encrypted-app-flash    Flash the encrypted app only.
  encrypted-flash        Flash the encrypted project.
  erase_flash            Erase entire flash chip.
  erase_otadata          Erase otadata partition.
  flash                  Flash the project.
  fullclean              Delete the entire build directory contents.
  menuconfig             Run "menuconfig" project configuration tool.
  monitor                Display serial output.
  partition_table        Build only partition table.
  partition_table-flash  Flash partition table only.
  read_otadata           Read otadata partition.
  reconfigure            Re-run CMake.
  set-target             Set the chip target to build.
  show_efuse_table       Print eFuse table.
  size                   Print basic size information about the app.
  size-components        Print per-component size information.
  size-files             Print per-source-file size information.
```

looks like your IDE isn't interfacing all them

WayneKeenan, 2019-10-13:
Sure, but not as CMake macros (of which I see 50+ in Clion)
I don’t want to spawn a shell script for the missing ones I mention, that doesn’t integrate so well when errors occur.

renzbagaporo, 2019-10-14:
Hi @WayneKeenan, this is similar to https://github.com/espressif/esp-idf/issues/2420, if I understand correctly. Due to limitations in invoking CMake targets, you need to have defined ESPPORT and ESPBAUD in your environment variables, however.

As for the clean target, CMake automatically creates a clean target that you can invoke from build system backend (`make clean` or `ninja clean` etc.). In CLion the option is in Build -> Clean.

WayneKeenan, 2019-10-14:
yes it is similar to #2420, adding monitor and erase_flash would be good, thanks.

WayneKeenan, 2019-10-18:
> Hi @WayneKeenan, this is similar to #2420, if I understand correctly. Due to limitations in invoking CMake targets, you need to have defined ESPPORT and ESPBAUD in your environment variables, however.

Indeed, hence I raised #4191

igrr, 2022-03-14:
The missing targets (monitor and erase_flash) were added in a1a2f3345909c717dd4c4e558b062e6272c3ae44, sorry for not updating the issue in time.

closed 2022-03-14 by igrr