espressif/esp-idf #4193: Can´t compile esp32-webserver

yoandresmza, 2019-10-12:
Hi,  I am trying to change the Wi-Fi mode of the FZ2707 (4 realys) module to connect it to my home Wi-Fi. I'm using W10.
I cloned the esp32-webserver and esp-idf repositories from github and installed the ESP tools but I can't build the project! Other projects like "hello_world" are well built.
When I try to build the project I have this error:

[image]

What am I doing wrong? Im new to this, any help would be important to me.
Thank you!

Andres

negativekelvin, 2019-10-13:
Which esp32-webserver project?

yoandresmza, 2019-10-13:
This one...
https://github.com/cyoung/esp32-webserver
Is a firmware for a 4 realy (module  FZ2707 from DIYMALL) based on esp-idf frameword

negativekelvin, 2019-10-13:
That project does not support cmake (idf.py), only make.  And it probably has only been tested on older versions of esp-idf maybe ≤3.2.  If you are asking for help with third party projects please open your issue on their GitHub repo or go to the esp32.com forum.

Alvin1Zhang, 2019-10-14:
@negativekelvin Thanks for the notes. @yoandresmza Thanks for reporting, for third party issues, please help create a post on our forum at esp32.com. Thanks.

closed 2019-10-14 by Alvin1Zhang