espressif/esp-idf #4194: fatal error: lwip/ip_addr.h: No such file or directory compilation terminated.

JasonLrh, 2019-10-14:
esp/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:24:26: fatal error: lwip/ip_addr.h: No such file or directory
esp/esp-idf/components/lwip/port/esp32/include/sys/socket.h:33:26: fatal error: lwip/sockets.h: No such file or directory

I use the latest source code but it is error even in building the basic program "Blink".
is there any solution?

Alvin1Zhang, 2019-10-14:
@JasonLrh Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

JasonLrh, 2019-10-14:
the earlier release I use works well on my dev-C board. now this release can't pass the compile . the error log is this title. thank s.

negativekelvin, 2019-10-14:
Did you `git submodule update --init --recursive`

freakyxue, 2020-01-17:
@JasonLrh 
Is there any other question

freakyxue, 2020-02-27:
I don't think it's a problem. If there is any problem, you can open it

Alvin1Zhang, 2020-02-28:
@JasonLrh Thanks for reporting. Feel free to reopen if the issue still happens. Thanks.

closed 2020-02-28 by Alvin1Zhang