espressif/esp-idf #4195: SerialBT.read() codepage question

lagunax, 2019-10-14:
Hi all.
using BluetoothSerial.h in project. trying to make BT-terminal, but stopped on encoding symbols.
Who can tell me where to find symbol-code-table?
sending 0123456789\n gives me strange codes with SerialBT.read():
```
symbol=code(dec)(int-value)
0=480
1=491
2=502
3=513
4=524
5=535
6=546
7=557
8=568
9=579
new line = 13 10 <- those codes are correct
```
in any codepages 0-9 numbers are 0x30-0x39 codes

Alvin1Zhang, 2019-10-14:
@lagunax Thanks for reporting. For "how" questions or third party issues, we suggest you could submit a post on our forum at https://esp32.com. Thanks.

lagunax, 2019-10-14:
eroor was in code, sorry, found my mistake

closed 2019-10-14 by lagunax