espressif/esp-idf #4197: Example esp-idf/examples/protocols/sockets/tcp_server accet statement

DrClaudio73, 2019-10-14:
##  Environment
- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (WroverKit): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-141-ga7e8d87d3
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
After first connection from a client the example can not accept further bind on socket.

//Detailed problem description goes here.
After first connection it is not possible to bind on the same socket.
the code with accept should be moved in the inner while (1)

Instead of:

static void tcp_server_task(void *pvParameters){
.....
while(1){
......
 listen_sock=socket();
 bind(listen_sock);
 listen(listen_sock);
 sock=accept(listen_sock);
 while(1){
    ......
   }             
   if (sock != -1) {
        ESP_LOGE(TAG, "Shutting down socket and restarting...");
        shutdown(sock, 0);
        close(sock);
   }
 }
}

I would go like this:

static void tcp_server_task(void *pvParameters){
.....
while(1){
......
 listen_sock=socket();
 bind(listen_sock);
 listen(listen_sock);
 while(1){
    sock=accept(listen_sock);
    ......
    if (sock != -1) {
        ESP_LOGE(TAG, "Shutting down socket and restarting...");
        shutdown(sock, 0);
        close(sock);
     }
  

 }             
 }
}

### Expected Behavior
I (3017) example: Socket created
I (3017) example: Socket bound, port 3333
I (3017) example: Socket listening
I (5667) example: Socket accepted
I (5667) example: Received 9 bytes from 192.168.1.101:
I (5667) example:  whatever....

E (5667) example: Shutting down socket and restarting...
I (12607) example: Socket accepted
I (12617) example: Received 9 bytes from 192.168.1.101:
I (12617) example:  whatever 2nd....

E (12617) example: Shutting down socket and restarting...

### Actual Behavior
I (3017) example: Socket created
I (3017) example: Socket bound, port 3333
I (3017) example: Socket listening
I (5667) example: Socket accepted
I (5667) example: Received 9 bytes from 192.168.1.101:
I (5667) example:  whatever....

I (6997) example: Connection closed
E (6997) example: Shutting down socket and restarting...
I (6997) example: Socket created
E (6997) example: Socket unable to bind: errno 112

### Steps to repropduce

1. build flash monitor example tcp_serve
2. echo -ne "whatever........\r\n" | nc -w1 192.168.1.103 3333
3. echo -ne "whatever 2nd........\r\n" | nc -w1 192.168.1.103 3333

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue
original code on repository tcp_server.c
```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

igrr, 2019-10-15:
Thanks, this has been resolved recently, the fix should be in master branch soon.

closed 2019-10-31 by DrClaudio73

AntumArk, 2020-03-30:
I have the same problem with tcp socket.
[image]

Once my client disconnects, or somekind of sending error occurs, socket fails to bind. I need to restart ESP to fix it.

DrClaudio73, 2020-03-30:
Did you try to move into the inner while(1){ } cicle the accept() statement? Also the final if (sock != -1) { } statement should be placed there.
More recent versions of esp-idf fix the issue.

AntumArk, 2020-03-30:
I have managed to make it work with moving while cycle according to master branch example. 

Now it is like this
while{
   Init address
   call socket function
   //Added from neweer exampele
  int opt = 1;
  setsockopt(listen_sock, SOL_SOCKET, SO_REUSEADDR, &opt, sizeof(opt));
///
   bind
   listen
   while{
       accept socket
        while{
             receive functions}
       catch socket error and shutdown socket.
}