espressif/esp-idf #4199: retain unused symbols

123swk123, 2019-10-15:
## Environment
- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-369-g4dac7c7df
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description
I have a design requirement which need to retain a function which would later be used by application through a function pointer based on the user choice.

```cpp

__used void someFunc(int arg1, int arg2);

void someFunc(int arg1, int arg2)
{
  /*some logics*/
}

typedef void (*myFun)(int arg1, int arg2);

void app_main()
{
  esp_partition_iterator_t part = esp_partition_find(ESP_PARTITION_TYPE_DATA, 0xEE, "usr_data");
  if (part == NULL)
  {
    ESP_LOGE(TAG, "unable to find the partition iterator");

    //TODO: show error in display

    return;
  }
  
  par = esp_partition_get(part);
  if (par == NULL)
  {
    ESP_LOGE(TAG, "unable to find the partition");  

    //TODO: show error in display

    return;
  }
  esp_partition_iterator_release(part);

  spi_flash_mmap_handle_t hPartMap;
  const uint8_t *dpUser;
  esp_partition_mmap(par, 0, gNwk_Para.par->size, SPI_FLASH_MMAP_DATA, (const void **)&dpUser, &hPartMap);

  myFun usrFunc = (myFun)(dpUser + SOME_OFFSET);
  usrFunc(...);
}

```

as per above example I would like to retain the function **someFunc**, I do use **__used** attribute for the compiler to emit the object code but due to linker **--gc-sections** option the emitted code section **.text.someFunc** is removed 

so do you have any workaround like by making linker fragment file or tweaking some of your build system to realize my requirement.

Note: I did manually add `KEEP(*(.text.someFunc))` in **esp32.project.ld.in** then I was successfull in retaining the symbol **someFunc**

mahavirj, 2019-10-15:
@123swk123 Adding `COMPONENT_ADD_LDFLAGS := -u <symbol_name>` to `component.mk` should help to address this

123swk123, 2019-10-15:
@mahavirj -u is suppose to remove the linker symbol definition so I don't think it will retain the symbol.
here is the description about -u option from linker manual.
`-u symbol
--undefined=symbol
Force symbol to be entered in the output file as an undefined symbol. Doing this may, for example, trigger linking of additional modules from standard libraries. -u may be repeated with different option arguments to enter additional undefined symbols. This option is equivalent to the "EXTERN" linker script command.`

right now only option that works for me is using KEEP(...) in ld file. so i have modified the ldgen.py utility to support modified linker fragment syntax which will insert KEEP(..) definition in the 
 corresponding section of project ld file.

here is my modified linker fragment syntax:

```
[mapping:main]
archive: libmain.a
entries:
  main:someFunc+ (default)
```
the `+` after the symbol **someFunc** tells ldgen to insert the KEEP(..) syntax.

123swk123, 2019-10-15:
here is the patch for ldgen

``` patch
diff --git a/tools/ldgen/fragments.py b/tools/ldgen/fragments.py
index 0339f5849..6331999a4 100644
--- a/tools/ldgen/fragments.py
+++ b/tools/ldgen/fragments.py
@@ -234,7 +234,7 @@ class Sections(Fragment):
         else:
             if sections_entry.endswith("+"):
                 section = sections_entry.replace("+", ".*")
-                expansion = section.replace(".*", "." + symbol)
+                expansion = section.replace(".*", "." + symbol.replace('+', ''))
                 return (section, expansion)
             else:
                 return (sections_entry, None)
@@ -306,6 +306,12 @@ class Mapping(Fragment):
                     pass
 
                 self.entries.add((obj, symbol, scheme))
+        elif key == 'retain':
+            for result in parse_results:
+                try:
+                    self.retain.add(result["retain"])
+                except KeyError:
+                    pass
 
     def get_key_grammars(self):
         # There are three possible patterns for mapping entries:
@@ -313,7 +319,7 @@ class Mapping(Fragment):
         #       obj (scheme)
         #       * (scheme)
         obj = Fragment.ENTITY.setResultsName("object")
-        symbol = Suppress(":") + Fragment.IDENTIFIER.setResultsName("symbol")
+        symbol = Suppress(":") + Word(alphas + "_", alphanums + "_" + "+.").setResultsName("symbol")
         scheme = Suppress("(") + Fragment.IDENTIFIER.setResultsName("scheme") + Suppress(")")
 
         pattern1 = obj + symbol + scheme
@@ -329,7 +335,6 @@ class Mapping(Fragment):
 
         return grammars
 
-
 class DeprecatedMapping():
     """
     Encapsulates a mapping fragment, which defines what targets the input sections of mappable entties are placed under.
diff --git a/tools/ldgen/generation.py b/tools/ldgen/generation.py
index 686defc3a..98bbd25ca 100644
--- a/tools/ldgen/generation.py
+++ b/tools/ldgen/generation.py
@@ -202,6 +202,9 @@ class PlacementRule():
         else:
             rule_string = "*%s:%s(%s)" % (archive, obj, sections_string)
 
+        if self.symbol and self.symbol.endswith('+'):
+            rule_string += "\n    KEEP(*(.text.%s))" % (sections_string)
+
         return rule_string
 
     def __eq__(self, other):
@@ -247,6 +250,7 @@ class PlacementRule():
         raise StopIteration
 
 
+import inspect
 class GenerationModel:
     """
     Implements generation of placement rules based on collected sections, scheme and mapping fragment.
@@ -536,7 +540,6 @@ class TemplateModel:
             try:
                 indent = member.indent
                 rules = member.rules
-
                 for rule in rules:
                     generated_line = "".join([indent, str(rule), '\n'])
                     output_file.write(generated_line)

```

mahavirj, 2019-10-17:
> @mahavirj -u is suppose to remove the linker symbol definition so I don't think it will retain the symbol.

As you rightly suggested that there are multiple ways to achieve this, but I was trying to propose something that can be achieved at component level itself. Above approach should work as well, you might want to try `hello_world` example from feature branch at https://github.com/mahavirj/esp-idf/commits/feature/keep_undefined_symbols (Here is relevant commit, https://github.com/mahavirj/esp-idf/commit/352e58c78f35e011298ef999351f6d720994db35)

igrr, 2019-10-17:
> @mahavirj -u is suppose to remove the linker symbol definition so I don't think it will retain the symbol.

Possibly you are misinterpreting the explanation of this flag in the LD manual. It says "undefine" a symbol, which doesn't mean "remove symbol definition". It actually means "add symbol to the list of undefined symbols". As the linker iterates over the object files inside a static library, it checks if the object file resolves references to any of the previously undefined symbols. If the object file doesn't resolve any such references, then it is discarded as whole.

We use `-u symbolname` in multiple places in IDF to force linking of some of the symbols, for example:

https://github.com/espressif/esp-idf/blob/2e6398affaeeac2f7ce40457a881f2dda57ad11f/components/esp32/component.mk#L22
https://github.com/espressif/esp-idf/blob/2e6398affaeeac2f7ce40457a881f2dda57ad11f/components/cxx/CMakeLists.txt#L16
https://github.com/espressif/esp-idf/blob/2e6398affaeeac2f7ce40457a881f2dda57ad11f/components/newlib/CMakeLists.txt#L25-L28

So i think the approach should also work in your case.

123swk123, 2019-10-17:
I have tried following both ways but still the symbol was not present in the final elf file and map file

option 1
```
target_link_options(${PROJECT_NAME}.elf PRIVATE -u .text.neo_led_spipixel_surface_fill)
```

option 2
```
target_link_options(${PROJECT_NAME}.elf PRIVATE -u neo_led_spipixel_surface_fill)
```

123swk123, 2019-10-17:
> I have tried following both ways but still the symbol was not present in the final elf file and map file
>
> option 1
>
> ```
> target_link_options(${PROJECT_NAME}.elf PRIVATE -u [...]

that's strange, this works in linux but not in windows!, originally I used this approach in cmake file but since it did not produce the expected result I started to workaround by patching the ldgen.

123swk123, 2019-10-17:
when i use
```
target_link_options(${PROJECT_NAME}.elf PRIVATE -u neo_led_spipixel_surface_fill)
```
it appends **-u neo_led_spipixel_surface_fill** to **LINK_FLAGS** in build.ninja but it does not insert the symbol to final elf

when i use
```
target_link_libraries(${COMPONENT_LIB} PRIVATE -u neo_led_spipixel_surface_fill)
```
it appends **-u neo_led_spipixel_surface_fill** to **LINK_LIBRARIES** in build.ninja but it does insert the symbol to final elf

this behavior is in linux, let me check in windows and get back.

mahavirj, 2019-10-23:
@123swk123 Do you have any update on above?

mahavirj, 2019-10-30:
Closing due to lack of response. Please feel to reopen in case issue still exists.

closed 2019-10-30 by mahavirj