espressif/esp-idf #4200: esp-idf-tools-setup-2.1 installation fails on windows10

moslef, 2019-10-15:
## Environment

- Operating System:     Windows 10 64bit

## Problem Description

Hello,

i downloaded the esp-idf-tools-setup-2.1 to install esp-idf on a windows 10 machine, but the installation fails every time. i dont understand the error description in the log output:

[Setup Log 2019-10-15 #006.txt](https://github.com/espressif/esp-idf/files/3729132/Setup.Log.2019-10-15.006.txt)

### Steps to repropduce

1. download esp-idf-tools-setup-2.1 from https://dl.espressif.com/dl/esp-idf-tools-setup-2.1.exe on windows 10 64bit
2. install esp-idf-tools-setup-2.1

## Debug Logs

[Setup Log 2019-10-15 #006.txt](https://github.com/espressif/esp-idf/files/3729143/Setup.Log.2019-10-15.006.txt)

Please tell what i can try to have a correct installation.

igrr, 2019-10-15:
Hi @moslef, thanks for reporting this. It looks like at the moment when you were trying to install IDF, cffi package has released a new version to Pypi, but did not have the wheels (binary dependencies) uploaded yet. Because of that, `pip` has tried to build the package from source, and failed.
It looks like the packages have been uploaded 2 hours ago: https://pypi.org/project/cffi/#files.
If you retry the installation, it should succeed this time.

moslef, 2019-10-16:
You are right, thank you. I close the issue.

closed 2019-10-16 by moslef