espressif/esp-idf #4203: ESP32 High/Low GPIO Interrupts crash

ExtremeGTX, 2019-10-15:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version: v4
- Module or chip used:  ESP32-WROOM-32D
- IDF versionv: 4.1-dev-437-gd2ad0f077
- Build System:         Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description
Trying to get `GPIO_INTR_HIGH_LEVEL` or `GPIO_INTR_LOW_LEVEL` working

### Expected Behavior
counter count up
### Actual Behavior
Crash `Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)`
### Steps to repropduce

1. open the gist link below
2. compile the code using esp-idf
3. flash the board

Connect jumper between Pin `25` and `3V3`

### Code to reproduce this issue
https://gist.github.com/ExtremeGTX/b7c135afdbeccdc12c66e4b7dde433eb
## Debug Logs

```I (12) boot: ESP-IDF v4.1-dev-437-gd2ad0f077 2nd stage bootloader
I (12) boot: compile time 19:11:39
I (13) boot: Enabling RNG early entropy source...
I (17) boot: SPI Speed      : 40MHz
I (22) boot: SPI Mode       : DIO
I (26) boot: SPI Flash Size : 4MB
I (30) boot: Partition Table:
I (33) boot: ## Label            Usage          Type ST Offset   Length
I (40) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (48) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (55) boot:  2 factory          factory app      00 00 00010000 00100000
I (63) boot: End of partition table
I (67) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x06794 ( 26516) map
I (86) esp_image: segment 1: paddr=0x000167bc vaddr=0x3ffb0000 size=0x02098 (  8344) load
I (89) esp_image: segment 2: paddr=0x0001885c vaddr=0x40080000 size=0x00400 (  1024) load
I (94) esp_image: segment 3: paddr=0x00018c64 vaddr=0x40080400 size=0x073ac ( 29612) load
I (115) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x1333c ( 78652) map
I (144) esp_image: segment 5: paddr=0x0003335c vaddr=0x400877ac size=0x02184 (  8580) load
I (154) boot: Loaded app from partition at offset 0x10000
I (154) boot: Disabling RNG early entropy source...
I (155) cpu_start: Pro cpu up.
I (159) cpu_start: Application information:
I (164) cpu_start: Project name:     hello-world
I (169) cpu_start: App version:      1
I (173) cpu_start: Compile time:     Oct 15 2019 19:11:36
I (180) cpu_start: ELF file SHA256:  0e605b27448b5c01...
I (186) cpu_start: ESP-IDF:          v4.1-dev-437-gd2ad0f077
I (192) cpu_start: Starting app cpu, entry point is 0x40080fec
I (192) cpu_start: App cpu up.
I (203) heap_init: Initializing. RAM available for dynamic allocation:
I (209) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (215) heap_init: At 3FFB3090 len 0002CF70 (179 KiB): DRAM
I (222) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (228) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (234) heap_init: At 40089930 len 000166D0 (89 KiB): IRAM
I (241) cpu_start: Pro cpu start user code
I (259) spi_flash: detected chip: generic
I (259) spi_flash: flash io: dio
I (260) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (268) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:5 
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400824ef  PS      : 0x00050034  A0      : 0x3ffb3280  A1      : 0x3ffb0690  
A2      : 0x00002040  A3      : 0x3ffb3340  A4      : 0x000637ff  A5      : 0x3ffb0670  
A6      : 0x3ff44000  A7      : 0x00000000  A8      : 0x3ffaffc4  A9      : 0x40087198  
A10     : 0x00000000  A11     : 0x3ffaffe8  A12     : 0x3ffb6c10  A13     : 0x00000000  
A14     : 0x02000000  A15     : 0x3ffb6c94  SAR     : 0x00000007  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x4000bff0  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400824ef

ELF file SHA256: 0e605b27448b5c016b8974f9d09dbe4526f0b0d54401993f3e596994a6030ac4

Backtrace: 0x400824ec:0x3ffb0690 0x3ffb327d:0x3ffb3340 |<-CORRUPTED

Core 1 register dump:
PC      : 0x400e3346  PS      : 0x00060634  A0      : 0x800d1d8a  A1      : 0x3ffb5d50  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d18ba  A9      : 0x3ffb5d20  
A10     : 0x00000000  A11     : 0x00060623  A12     : 0x00060620  A13     : 0x00000001  
A14     : 0x00060620  A15     : 0x3ffb6c70  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 0e605b27448b5c016b8974f9d09dbe4526f0b0d54401993f3e596994a6030ac4

Backtrace: 0x400e3343:0x3ffb5d50 0x400d1d87:0x3ffb5d70 0x40088a62:0x3ffb5d90 0x40086f71:0x3ffb5db0

Rebooting...

```

negativekelvin, 2019-10-15:
GPIO_INTR_HIGH_LEVEL will keep generating interrupts as long as input is high and CPU will be blocked.

ExtremeGTX, 2019-10-16:
pardon my ignorance.
if CPU is Blocked forever then how to use this kind of interrupt.
i tried to call `gpio_set_intr_type(GPIO_INPUT_IO_0, GPIO_INTR_DISABLE);` from ISR to disable the interrupts, also didn't work.

I lowered the priority to `gpio_install_isr_service(ESP_INTR_FLAG_LEVEL1);` and called `gpio_set_intr_type(GPIO_INPUT_IO_0, GPIO_INTR_DISABLE);` from CPU1 also didn't work.

but i tried to use both cores, one to handle the interrupt and the other just `printf("alive")`, but both cores stall and i get endless reset.

also tried this assembly example [https://github.com/darthcloud/esp32_highint5_gpio], didn't work

My question is How to use this interrupt, is there a working example ?

costaud, 2019-10-28:
Sorry I can't open the link to your code. But according to my test, I think the problem here could be that the interrupt get triggered immediately after you called gpio_install_isr_service. Since GPIO25 is already connected with v3.3. 

So the interrupt handler got no chance to run at all. 

The solution is to set intr type to GPIO_INTR_DISABLE during the initialization, and after everything is done and called, set the intr type to high or low level. 

In the interrupt handler, you can set the intr type to disable or to the opposite level.

ExtremeGTX, 2019-11-04:
The problem was EDGE/LEVEL interrupt assignment.

closed 2019-11-04 by ExtremeGTX