espressif/esp-idf #4204: Unable to connect to Wi-Fi AP if its SSID contains an apostrophe

MelvYoto, 2019-10-16:
## Environment

- Development Kit:      [ESP32-LyraT]
- Kit version (for WroverKit/PicoKit/DevKitC): []
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.3
    // v4.1-dev-437-gd2ad0f077
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a5
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

Unable to connect to an access point if its SSID contains an apostrophe character (0x27) e.g. phone hotspot with a name in the form: Jim's iPhone

### Expected Behavior

Should be able to connect to APs with apostrophe(s) in their SSID.

I tried escaping the apostrophe in the example code, but none of the following worked:
- Jim's iPhone
- Jim\\'s iPhone
- Jim\047s iPhone

Note: after going into the iPhone's settings and changing its name from:
- Jim's iPhone
To:
- Jims iPhone

The station example is now able to connect for this configuration. However, we can't expect customers to remove apostrophe's from their SSIDs.

Also note: the iPhone appears to be using the correct character for it's single apostrophe (i.e. 0x27):

$ **netsh wlan show profiles**
    All User Profile     : Jim's iPhone

$ **netsh wlan show profiles | od -x**
000000 2020 2020 6c41 206c 7355 7265 5020 6f72
0000020 6966 656c 2020 2020 3a20 4a20 6d69 7327
0000040 6920 6850 6e6f 0a65

(where 7327 = 's)

Finally, note that although connection fails, esp_wifi_scan_get_ap_records() is able to correctly list the SSID with the apostrophe e.g.:

                            SSID | Channel | RSSI |    Auth Mode
----------------------------------------------------------------
                    Jim’s iPhone |       6 |  -44 |     WPA2 PSK

### Actual Behavior

Fails to connect when there is an apostrophe in the SSID, which manifests as event SYSTEM_EVENT_STA_DISCONNECTED, code 201, no AP found. The example code retries multiple times without success.

### Steps to repropduce

Configure examples/wifi/getting-started/station to attempt to connect to an SSID with an apostrophe in the name.

### Code to reproduce this issue

See steps to reproduce: only the example configuration is changed (i.e. SSID / Password)

## Debug Logs

**From v3.3**

I (1091) wifi station: connect to ap SSID:Jim's iPhone password:abcdefghijkl
D (3141) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:Jim's iPhone, ssid_len:12, bssid:00:00:00:00:00:00, reason:201,wifi reason: no ap found
V (3141) event: enter default callback
V (3141) tcpip_adapter: check: local, if=0 fn=0x4011c238
0x4011c238: tcpip_adapter_down_api at C:/Users/jim/esp/test/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1098
D (3151) tcpip_adapter: if0 start ip lost tmr: enter
D (3151) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffc5808 interval=120 ip=0
V (3161) tcpip_adapter: call api in lwip: ret=0x0, give sem
V (3171) tcpip_adapter: check: remote, if=0 fn=0x4011c238
0x4011c238: tcpip_adapter_down_api at C:/Users/jim/esp/test/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1098
V (3181) esp_adapter: thread sem get: sem=0x3ffc58fc
V (3181) event: exit default callback
V (3181) esp_adapter: thread sem get: sem=0x3ffc58fc
I (3191) wifi station: retry to connect to the AP
I (3191) wifi station: connect to the AP fail

**From esp-idf v4.1-dev-437-gd2ad0f077**

I (2153) wifi station: connect to ap SSID:Jim's iPhone password:abcdefghijkl:wq
D (4203) event: running post WIFI_EVENT:5 with handler 0x400dc074 on loop 0x3ffbe6c0
0x400dc074: handle_sta_disconnected at C:/Users/jim/esp/test/esp-idf/components/tcpip_adapter/event_handlers.c:56
V (4203) tcpip_adapter: check: local, if=0 fn=0x400dcdc8
0x400dcdc8: tcpip_adapter_down_api at C:/Users/jim/esp/test/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1146
D (4203) tcpip_adapter: if0 start ip lost tmr: enter
D (4213) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffc5870 interval=120 ip=0
V (4223) tcpip_adapter: call api in lwip: ret=0x0, give sem
V (4223) tcpip_adapter: check: remote, if=0 fn=0x400dcdc8
0x400dcdc8: tcpip_adapter_down_api at C:/Users/jim/esp/test/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:1146
V (4233) esp_adapter: thread sem get: sem=0x3ffba174
D (4233) event: running post WIFI_EVENT:5 with handler 0x400d5428 on loop 0x3ffbe6c0
0x400d5428: event_handler at c:\users\jim\esp\test\esp-idf\examples\wifi\getting_started\station\build/../main/station_example_main.c:44
V (4243) esp_adapter: thread sem get: sem=0x3ffba174
I (4253) wifi station: retry to connect to the AP
I (4253) wifi station: connect to the AP fail

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

esp-HarveyRong, 2019-11-07:
Hi @MelvYoto Thank you very much for your feedback, I will debug the problem and update it in time.

esp-HarveyRong, 2019-11-13:
Hi @MelvYoto I think it may be a problem with the input method. 
I found that the Jim's iPhone hexadecimal of the iphone hotspot ssid is 0x4A, 0x69, 0x6D, **0xE2, 0x80, 0x99,** 0x73, 0x20, 0x69, 0x50, 0x68, 0x6f, 0x6e, 0x65.
In fact, Jim's iPhone's **correct hexadecimal** is Ox4A, 0x69, 0x6D, **0x27,** 0x73, 0x20, 0x69, 0x50, 0x68, 0x6f, 0x6e, 0x65.
So if you use 0x4A, 0x69, 0x6D, 0xE2, 0x80, 0x99, 0x73, 0x20, 0x69, 0x50, 0x68, 0x6f, 0x6e, 0x65, you can connect the iphone hotspot to Jim's iPhone ssid.

Alvin1Zhang, 2020-01-15:
@MelvYoto Thanks for reporting, would you please help share if any updates for this issue? Thanks.

MelvYoto, 2020-01-15:
Hi @Alvin1Zhang - thanks very much for your feedback, I was previously relying on the output from 'netsh wlan show profiles' for debugging, but Windows is mapping to its own character encoding which masked the problem. It looks like the issue is a result of Apple introducing Smart Punctuation which changes the standard ASCII quote into a right-facing quote (U+2019, or UTF-8 E2 80 99). As you suggest, one solution is to remap the character. Another is for the user to turn off iOS Smart Punctuation, which results in the standard 0x27 quote character being used by the iOS device.

closed 2020-01-15 by MelvYoto