espressif/esp-idf #4207: Cannot use GPIO36 for Bootloader Control of Test Mode or Factory Reset

bgintz-rsc, 2019-10-17:
I need to implement the bootloader support for Factory Reset and Test mode entry via two switches.  (Entered by holding the input low for specified period of Time while power is applied as configured in sdkconfig file as shown below). 

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=3
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V=
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
CONFIG_BOOTLOADER_FACTORY_RESET=y
CONFIG_BOOTLOADER_NUM_PIN_FACTORY_RESET=39
CONFIG_BOOTLOADER_OTA_DATA_ERASE=y
CONFIG_BOOTLOADER_DATA_FACTORY_RESET="nvs"
CONFIG_BOOTLOADER_APP_TEST=y
CONFIG_BOOTLOADER_NUM_PIN_APP_TEST=36
CONFIG_BOOTLOADER_HOLD_TIME_GPIO=15
CONFIG_BOOTLOADER_WDT_ENABLE=y
CONFIG_BOOTLOADER_WDT_TIME_MS=60000
CONFIG_BOOTLOADER_OTA_ROLLBACK=y
CONFIG_BOOTLOADER_OTA_NO_FORCE_ROLLBACK=y

Checking the bootloader messages confirms that test mode will not be entered when driven by GPIO36.  (The device immediately boots into the production code). However the Factory reset will work using GPIO39.  

 Swapping modes between the two GPIO's will validate that the mode is correctly configured (i.e. Test mode will now work but fact config won't.  The two switches are wired identically on the board.

KonstantinKondrashov, 2019-10-17:
Hi @bgintz-rsc!
There was a note - `Note that GPIO34-39 do not have an internal pullup and an external one must be provided.`
I am not sure what is special with GPIO36. Could you confirm that it is really low while a start?
One more point, is GPIO36 used in your app and how is it setting (analog)?
Would you please provide more details as suggested in the issue template? IDF version, board ...

Thanks.

bgintz-rsc, 2019-10-17:
We do have pull-ups on both GPIO36 and GPIO39.  The button is used in the application to start a motor and we have no issues using it there.  It is set up as an input.  No analog functions.

bgintz-rsc, 2019-10-17:
Thanks Konstantin - 

Please See attached images of power up with each of the two buttons pressed. 

Yellow = 3.3V
Blue = GPIO36
Pink = GPIO39

It appears that something is pulling GPIO36 high (blue in the second photo) for about 600ms.  This is not seen on the second photo.  We can find nothing in our hardware that would account for this. 
[image]
[image]

Best 
Bobby

stickbreaker, 2019-10-18:
@bgintz-rsc your diagnostics and results point to a hardware problem on GPIO36.  I'd have to see a schematic and the actual article to confirm. 

Chuck.

bgintz-rsc, 2019-10-19:
Thanks for responding Chuck.  

Here is the schematic of the wiring for the two buttons.  They are identical.  There is nothing else connected to GPIO 36 and GPIO 39 on the board (per the schematic).   One of them works and the other does not.  There is nothing I could find on our board holding GPIO 36 high on reset as seen on the scope.  I suppose I can test with a bare ESP32 module (ESP32-DevKitC).  Please clarify - Have you successfully tried to use GPIO36 for factory reset or Test mode or is your assessment from the context of this thread?  It would be good to know if someone else has tested this and it worked.  

[image]

stickbreaker, 2019-10-19:
The reason I think you have a hardware problem is that the [ESP32 Hardware Reference V4.0](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) page 44 state:
>**4. IO_MUX and GPIO Matrix**
*4.1Overview*
The ESP32 chip features 34 physical GPIO pads. Each pad can be used as a general-purpose I/O, or be connected to an internal peripheral signal. The IO_MUX, RTC IO_MUX and the GPIO matrix are responsible for routing signals from the peripherals to GPIO pads. Together these systems provide highly configurable I/O.Note that the I/O GPIO pads are 0-19, 21-23, 25-27, 32-39, while the output GPIOs are 0-19, 21-23, 25-27,32-33. **_GPIO pads 34-39 are input-only._**

Based on your scope capture, something is driving GPIO36.

Chuck.

negativekelvin, 2019-10-20:
There is already a hardware bug with 36&39 because they are connected to the LNA so possibly something at startup with 36 too

bgintz-rsc, 2019-10-20:
Thanks for pointing this out negativekelvin.  I found reference to an issue with GPIO36 and 39 below but not the LNA issue you describe.  Do you have a reference? 

https://www.espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf

[image]

negativekelvin, 2019-10-20:
That is the one I meant

stickbreaker, 2019-10-20:
@negativekelvin can you imagine any way that these described issues would generate voltage on the gpio pin that his Oscilloscope would see?

negativekelvin, 2019-10-21:
Not really and I just tested gpio36 on a wrover kit board and it works to detect factory reset condition with 10k pullup, 4.7k series (values I had)

bgintz-rsc, 2019-10-21:
Thanks for the info on the test **negativekelvin**! 

What version of the idf were you using?  Would you be willing to share your bootloader folder (or at least bootloader.bin) for me to attempt to replicate this on our board?  We are using AmazonFreeRTOS and our boot message for the esp-idf shows:  **v_3.1.5-105-g7313c836a5_**

Also,  I agree **stickbreaker** that the errata doe not explain what I am seeing.  I will look at the PCN layout (done by someone else) for a possible unintended bridge.  But think that is unlikely with the checks already done vis-a-vis the schematic.

bgintz-rsc, 2019-10-22:
negativekelvin

May I get a copy of the bootloader.bin file you tested this with please?

bgintz-rsc, 2019-11-04:
Update: we have done a deep dive on our HW and still see nothing that could be pulling the GPIO36 line high.  Note:  we are using the 8MB part and I believe that you tested on the standard 4M VROOM.

negativekelvin, 2019-11-05:
How many units have you tested?  I trashed the project after test, but it was just the standard esp-idf 4.1 bootloader.

bgintz-rsc, 2019-11-11:
We have tested on multiple units. It is very repeatable.

KonstantinKondrashov, 2020-05-13:
Hi @bgintz-rsc !
Do you have progress with this issue? I still can not get your result, it works for me.
Thanks.

closed 2021-02-26 by Alvin1Zhang