espressif/esp-idf #4209: Ethernet OTA error when image file size is larger than hollow-word.bin

aa1622296203, 2019-10-17:
## Environment

- Development Kit:      [ESP32-Ethernet-kit-A]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1.1]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it): 
     v4.0-beta1-180-ga21eb04cc
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

Hi

I added OTA features to my esp32 ethernet project and have a problems I have installed the Ethernet driver correctly and got the local IP (DHCP succeeded)): 

When I use Hello word.bin (file size is 150kb), I can successfully implement OTA(Almost every test was a success), however when I use my own project's .bin file (file size is 1.2MB),  OTA  always failed and  always return the errors :

```cpp
[15:05:53:726] <0x1b>[0;32mI (1085052) esp_https_ota: Starting OTA...<0x1b>[0m␍␊
[15:05:53:726] <0x1b>[0;32mI (1085052) esp_https_ota: Writing to partition subtype 17 at offset 0x200000<0x1b>[0m␍␊
[15:06:33:788] <0x1b>[0;31mE (1125122) esp-tls: read error :-76:<0x1b>[0m␍␊
[15:06:33:804] <0x1b>[0;31mE (1125122) TRANS_SSL: esp_tls_conn_read error, errno=No more processes<0x1b>[0m␍␊
[15:06:38:860] <0x1b>[0;32mI (1130182) esp_https_ota: Connection closed, all data received<0x1b>[0m␍␊
[15:06:38:860] <0x1b>[0;32mI (1130182) esp_image: segment 0: paddr=0x00200020 vaddr=0x3f400020 size=0x33b7c (211836) map<0x1b>[0m␍␊
[15:06:38:988] <0x1b>[0;32mI (1130302) esp_image: segment 1: paddr=0x00233ba4 vaddr=0x3ffbdb60 size=0x03858 ( 14424) <0x1b>[0m␍␊
[15:06:38:988] <0x1b>[0;32mI (1130312) esp_image: segment 2: paddr=0x00237404 vaddr=0x40080000 size=0x00400 (  1024) <0x1b>[0m␍␊
[15:06:39:004] <0x1b>[0;32mI (1130312) esp_image: segment 3: paddr=0x0023780c vaddr=0x40080400 size=0x08804 ( 34820) <0x1b>[0m␍␊
[15:06:39:020] <0x1b>[0;31mE (1130342) esp_image: invalid segment length 0xffffffff<0x1b>[0m␍␊
[15:06:39:020] <0x1b>[0;31mE (1130342) main: Firmware Upgrades Failed 5379, 0x00001503<0x1b>[0m␍␊
[15:06:39:036] <0x1b>[0;31mE (1130342) main: ota 1st fail!,again<0x1b>

```
Same hardware, same server, different image file sizes,the small image can succeed, but the larger image is not successful, so I guess this is not the HTTP server fault, but the defect of ESP IDF. 
	
I made more tests to test my conjecture, to switch the network mode to WiFi STA, retest the OTA, and compare the two images:

- hollow-word.bin(150KB)   Test 5 times, successful 5 .
- myself.bin(1.2MB)   Test 5 times, successful 1 .

I'd like to know if you have tested the OTA when the firmware size is larger than hello word.bin (for example, the image size is larger than 1.5 MB).In particular, have you tested the OTA situation when using ip101(ethernet mode) for network communication?

aa1622296203, 2019-10-18:
HI,

I have solved this problem. There should be a problem with the default settings of ESP IDF.

Thanks！

closed 2019-10-18 by aa1622296203

projectgus, 2019-10-18:
Hi @aa1622296203 ,

Glad you found a fix.

> There should be a problem with the default settings of ESP IDF.

Can you explain what setting you had to change to make this work?

We have seen reports of similar errors when the server truncated the file (ie did not send all bytes).

Angus

aa1622296203, 2019-11-19:
Use larger DMA buffer,default is 512 Bytes ,change to 2048,itworks well

aa1622296203, 2019-11-27:
not 2048, 1600 is correct

BuddhikaNish, 2020-10-06:
Hi @aa1622296203 , 
can you explain how to change the DMA buffer size using arduino.

suffiyanshaikh, 2021-09-29:
i am also trying to doing ota using ethernet its working with wifi but not working with ethernet, below are the logs can you suggest some soultion

E (395143) esp-tls: couldn't get hostname for :beta.alitersolutions.com:
E (395143) esp-tls: Failed to open new connection
E (395153) TRANS_SSL: Failed to open a new connection
E (395153) HTTP_CLIENT: Connection failed, sock < 0
E (395163) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (395173) esp_https_ota: Failed to establish HTTP connection