espressif/esp-idf #4210: ESP32 CMake, How to add my own static library file  to project? (ESPIDF v4.1dev)

drony, 2019-10-17:
I added the following two lines to CMakeLists.txt

```
add_library(BCXConfiguration STATIC IMPORTED)
set_property(TARGET BCXConfiguration PROPERTY IMPORTED_LOCATION ${CMAKE_CURRENT_SOURCE_DIR}/libs/libBCXConfiguration.a)
```
[staticlib_helloworld.zip](https://github.com/espressif/esp-idf/files/3739176/staticlib_helloworld.zip)

**but when I compile it gives the following error**
```
undefined reference to `BakelorEDK::BCXConfiguration::BCXConfiguration()'
undefined reference to `BakelorEDK::BCXConfiguration::Init()'
....
...
```

i attached simple hello world sample app. thanks in advance

renzbagaporo, 2019-10-21:
You need to link the library to its "parent" component:

```
target_link_libraries(${COMPONENT_LIB} PRIVATE/PUBLIC/INTERFACE BCXConfiguration)
```

Alvin1Zhang, 2019-11-05:
@drony Thanks for reporting. Would you please help check whether your issue has been resolved? Thanks.

drony, 2019-11-19:
not resolved;
BCXConfiguraton.a static library used nvs_flash component inside. when  i use bcxconfiguration.a libraryin project. it give undefined reference error nvs_flash_init(), nvs_flash_erase() etc. 

--------------------------------------------------------------------

After build SomeSDK application, I copied the BCXConfiuration library from the following path
SomeSDK\build\esp-idf\BCXConfiguration\libBCXConfiguration.a

and i want use BCXLibrary standalone and i want to use with hellostatic app in attachment
[SomeSDK&HelloStatic.zip](https://github.com/espressif/esp-idf/files/3863311/SomeSDK.HelloStatic.zip)

renzbagaporo, 2019-11-21:
Done @drony.

closed 2019-11-21 by renzbagaporo