espressif/esp-idf #4214: Can't checkout release/v3.3

kostyan5, 2019-10-18:
After cloning esp-idf with:
```git clone --recurse-submodules https://github.com/odyn-ai/esp-idf.git```

and trying to checkout release/v3.3 branch with:
```git checkout release/v3.3```

 I get the following error:
```
fatal: not a git repository: ../../../.git/modules/components/aws_iot/aws-iot-device-sdk-embedded-C
fatal: could not reset submodule index
```

negativekelvin, 2019-10-19:
Try `git submodule update --init --recursive`

kostyan5, 2019-10-19:
That doesn't help either:

```
git checkout --recurse-submodules release/v3.3
fatal: not a git repository: ../../../.git/modules/components/aws_iot/aws-iot-device-sdk-embedded-C
fatal: could not reset submodule index
```

shubhamkulkarni97, 2019-11-05:
@kostyan5 I followed the steps as mentioned by you in the issue. It works fine for me. Can you please try following steps once?

1. Clone `esp-idf`
2. Run command `git submodule update --init --recursive`
3. Run command `git checkout release/v3.3`

Let me know if you face any issues.

shubhamkulkarni97, 2019-11-11:
@kostyan5 Do you have any updates?

kostyan5, 2019-11-11:
@shubhamkulkarni97 It's working now.  Something must have gotten fixed.  Thanks.

closed 2019-11-12 by mahavirj