espressif/esp-idf #4217: mDNS should be configurable via KConfig

atanisoft, 2019-10-19:
## Environment

- Development Kit:     ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v3
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v3.3 (all versions really)
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 5.2.0
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

mDNS task runs at a very low priority and pinned to core 0 (PRO_CPU). It would be good to make these options configurable via KConfig:
1. task priority
2. task pinned to core (0, 1, floating)
3. mdns_service_add timeout (used [here](https://github.com/espressif/esp-idf/blob/master/components/mdns/mdns.c#L4377))
4. expose MDNS_TIMER_PERIOD_US 

### Expected Behavior

Calling mdns_service_add should not timeout due to higher priority task running on the same core.

### Actual Behavior

mdns_service_add returns ESP_FAIL on first call as seen in https://github.com/atanisoft/ESP32CommandStation/issues/26. When the API returns ESP_FAIL I've added a workaround to reschedule the call to mdns_service_add at a later time which returns ESP_OK.

Even with mdns_service_add() returning ESP_OK the service description is not seen from other devices (rPi as example):
```
pi@RPi-JMRI:~/openmrn $ avahi-browse -a -t -k
+  wlan0 IPv6 _openlcb-can._tcp                             _openlcb-can._tcp    local
+  wlan0 IPv4 _openlcb-can._tcp                             _openlcb-can._tcp    local
+  wlan0 IPv6 RPI-JMRI                                      _device-info._tcp    local
+  wlan0 IPv4 RPI-JMRI                                      _device-info._tcp    local
+  wlan0 IPv6 RPI-JMRI                                      _smb._tcp            local
+  wlan0 IPv4 RPI-JMRI                                      _smb._tcp            local
```

### Steps to repropduce

in app_main() set the current task priority higher than 1 and not call vTaskDelay(), inside the code call mdns_service_add() and verify the return value.

### Code to reproduce this issue

https://github.com/atanisoft/ESP32CommandStation/tree/development can be used to reproduce this by using PlatformIO to compile/flash it to the ESP32 using the default configuration settings.

no1seman, 2019-10-19:
+1 and also need to make task stack size configurable

atanisoft, 2019-10-19:
> make task stack size configurable

I'm not certain it can be shrunk much unfortunately. There doesn't seem to be much need to increase it.  But if you have a usecase for adjusting it then I'm all for the extra flexibility, I just haven't found one yet.

no1seman, 2019-10-20:
@atanisoft,
as for now it defined: #define MDNS_SERVICE_STACK_DEPTH    4096 
in my case (4 registed services) I have never seen stack usage more than 50%:
[image]

atanisoft, 2019-10-20:
@no1seman makes sense, it could likely be dropped some in your case.

Can you share the code for generating the table with usage percent? I haven't been able to get the percentage to make much sense in my tests.

no1seman, 2019-10-20:
@atanisoft ,
it's impossible for now, because there's to much dependencies in my project, but it's not so hrd to make, just look into tskTCB structure ($IDF_PATH/components/freertos/tasks.c)  and you'll be able to make your own.

closed 2019-11-18 by espressif-bot