espressif/esp-idf #4218: Feature request: std::aligned_storage

PerMalmberg, 2019-10-19:
Hi,

Are there any plans to implement std::aligned_storage, it is part of C++11, after all?

Currently it results in the following error.

```
/home/permal/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/newlib/libnewlib.a(heap.c.obj):(.literal.memalign+0x0): undefined reference to `memalign_function_was_linked_but_unsupported_in_esp_idf'
/home/permal/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/newlib/libnewlib.a(heap.c.obj): in function `memalign':
/home/permal/code/esp-idf/components/newlib/heap.c:38: undefined reference to `memalign_function_was_linked_but_unsupported_in_esp_idf'
```

Br,

Per

igrr, 2019-10-19:
Hi @PerMalmberg, we do plan to add `memalign` in the context of the ESP32-S2 chip support, where certain alignment is required by some of the peripheral DMAs. What is your use case for the ESP32, are you porting some existing code? Thanks.

PerMalmberg, 2019-10-19:
I have a memory buffer I use for placement-new. I'd like to turn that into std::aligned_storage since I recently was told it is apparently undefined behaviour unless it is aligned. (The argumentation was somewhat weak and I can't find anything supporting it, so I'm currently just exploring this)

Here's the code for reference: https://github.com/PerMalmberg/Smooth/blob/master/lib/smooth/include/smooth/core/fsm/StaticFSM.h#L79

igrr, 2019-10-19:
Will this work for a buffer which is part of a structure/class? IIRC you can use `alignas(sizeof(State))` to align structure members.

PerMalmberg, 2019-10-19:
Oh, right, there's the alignas() keyword too. It's results in a bit more verbose code since it can't be applied directly to the buffer, but it  it does seem to work, though I'll have to try it out a bit more tomorrow.

```c++
struct alignas(StateSize) state_t
{
    uint8_t mem[2][StateSize]{};
} state;
```

Want to keep this issue open as a reference for others wondering the same thing or should I close it?

igrr, 2019-10-20:
Let's keep it open, will close when memalign is implemented.

PerMalmberg, 2019-10-20:
Ok, this is confusing.

It seems memalign is implemented, but only under certain circumstances. I've got two projects, one which builds a test project for my framework (linked above) and another project which builds an app using the framework. The test project builds and links just fine, the app builds but does not link, as per the first post.

Right now I can't tell what the difference is between these two. They use the same sdkconfig, same compile, same IDF  and they're both building from the same framework source.

The comment on [line 88 of heap.c](https://github.com/espressif/esp-idf/blob/6517c5033aab27afe491e433d6beeba93dfb1aa7/components/newlib/heap.c#L88) says memalign is not available in the heap-component, and newlib's implementation mustn't be linked in. Does that mean my test-project somehow links newlib anyhow since it does link (and runs just fine)?

igrr, 2019-10-20:
Possibly under some circumstances std::aligned_storage doesn't use memalign, or it gets somehow optimized out if the compiler can determine that alignment is satisfied at compile time? Looking at the map file can tell if memalign got linked in (and from which object file), and the disassembly of your function can reveal how the aligned_storage got implemented.

PerMalmberg, 2019-10-20:
From mqtt.map, the project that links successfully:
```
.literal.memalign
                0x0000000000000000        0x4 esp-idf/newlib/libnewlib.a(heap.c.obj)
 .text          0x0000000000000000        0x0 esp-idf/newlib/libnewlib.a(heap.c.obj)
 .data          0x0000000000000000        0x0 esp-idf/newlib/libnewlib.a(heap.c.obj)
 .bss           0x0000000000000000        0x0 esp-idf/newlib/libnewlib.a(heap.c.obj)
 .text.memalign
                0x0000000000000000        0xd esp-idf/newlib/libnewlib.a(heap.c.obj)
```
```
memalign                                          esp-idf/newlib/libnewlib.a(heap.c.obj)
memalign_function_was_linked_but_unsupported_in_esp_idf esp-idf/newlib/libnewlib.a(heap.c.obj)
```

This is from g3.map, the failing project:

```
.literal.memalign
                0x0000000040081034        0x4 esp-idf/newlib/libnewlib.a(heap.c.obj)

 .text.memalign
                0x0000000040095870        0xd esp-idf/newlib/libnewlib.a(heap.c.obj)
                0x0000000040095870                memalign
```
```
memalign                                          esp-idf/newlib/libnewlib.a(heap.c.obj)
                                                  /home/permal/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/esp32-psram/libstdc++.a(new_opa.o)
memalign_function_was_linked_but_unsupported_in_esp_idf esp-idf/newlib/libnewlib.a(heap.c.obj)
```

So in both cases it looks as if it doesn't actually link to an actual implementation of memalign (eventthough it points to newlib?) Does the adresses/offsets being 0 have a significance?

The following is from an objdump of mqtt.elf:

```
400e9351:	a4a032        	movi	a3, 164
400e9354:	373a      	add.n	a3, a7, a3
            struct alignas(StateSize) state_t
400e9356:	00a1c2        	movi	a12, 0x100
400e9359:	00a0b2        	movi	a11, 0
400e935c:	02d7a2        	addmi	a10, a7, 0x200
400e935f:	201110        	or	a1, a1, a1
400e9362:	a37e25        	call8	4008cb44 <memset>
            StaticFSM() = default;
```

As there's no call to memalign,  only to memset, it's been optimized out?

I'm trying to get a disassembly for g3 too, but since I don't have an elf file I need to get the compiler to do that.

uLipe, 2020-06-04:
Hi @PerMalmberg , there is an experimental branch with our new memory allocator, this branch also comes with `memalign` properly implemented, and using it you may able to use the `std::aligned_storage`.

Check it out:
https://github.com/espressif/esp-idf/tree/feature/freertos_upstream_plus_tlsf_allocator_preview

We are planning to merge this branch into master soon.

PerMalmberg, 2020-06-05:
Thanks for the update. You'll probably have merged this before I have chance to get around to trying it out.

igrr, 2022-03-14:
std::aligned_storage is supported in v4.3 and later releases.

closed 2022-03-14 by igrr