espressif/esp-idf #4220: Wifi doesn't work since yesterday

Jupp23, 2019-10-20:
## Environment

- Development Kit:      NodeMCU ESP32
- Kit version (for WroverKit/PicoKit/DevKitC): 
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-475-gb13ba57ca
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     Windows
- Power Supply:         USB xor external 5V

## Problem Description

since yesterday, i can compile w/o problems, but after flashing wifi won't start. Even if i use the unchanged example "wifi/getting_started/station". See verbose bootlog:
[ESP32-ErrorMessage.txt](https://github.com/espressif/esp-idf/files/3748081/ESP32-ErrorMessage.txt)

There is the Message
net80211: OS adapter function version error! Version 3 is expected, but it is 4
near end of file, followed by
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4008b654

As mentioned above, i use the unchanged example code.

### Expected Behavior
wifi should work
### Actual Behavior
wifi doesn't work
### Steps to repropduce

1. compile station example
2. flash station example

negativekelvin, 2019-10-20:
Did you update your submodules?

Jupp23, 2019-10-21:
yes, also used a fresh installation of idf. If this is not normal behaviour, then it seems, an old artifact of the V3.2 IDF is somewhere in my system and is taken for linking.

Jupp23, 2019-10-21:
Seems to be a problem with V ESP-IDF v4.1-dev-475-gb13ba57ca. This Version is used for 2nd stage bootloader too. Falling back to v4.1-dev-474-g2e6398aff giving a 2nd stage bootloader ESP-IDF v3.3 and "cpu_start: ESP-IDF: ESP-IDF v4.1-dev-474-g2e6398aff.
Now Code works as before.

negativekelvin, 2019-10-21:
Falling back? 2e6398a is the latest commit in master.

Jupp23, 2019-10-21:
hm... So, where did i get the other version (s.o.) from? Some days before i did a git pull and git submodule update --init --recursive as mentioned in the manual... strange :-(