espressif/esp-idf #4221: LWIP TCP/IP sockets still working after WIFI disconnect event

Patrik-Berglund, 2019-10-21:
## Environment
- Module or chip used: ESP32-WROVER
- IDF version v3.3

## Problem Description
We have seen that it's possible to successfully open and use a TCP socket after the WIFI disconnect event has been raised.

The TCP traffic can be up and working for days after the "SYSTEM_EVENT_STA_DISCONNECTED" was rasied without any debug output from the wifi subsystem.

```
[0;32mI (2035) WIFI: SYSTEM_EVENT_STA_CONNECTED.[0m
[0;32mI (4695) event: sta ip: 192.168.0.134, mask: 255.255.255.0, gw: 192.168.0.1[0m
[0;32mI (4695) WIFI: **SYSTEM_EVENT_STA_GOT_IP.[0m**

**Later followed by**

[0;32mI (18044695) event: station ip lost[0m
[0;32mI (18044695) WIFI: SYSTEM_EVENT_STA_LOST_IP.[0m
[0;32mI (18044695) WIFI: SYSTEM_EVENT_STA_DISCONNECTED.[0m
```

freakyxue, 2019-11-13:
hi @Patrik-Berglund 
We will debug the problem as soon as possible

freakyxue, 2019-11-15:
@Patrik-Berglund
 When I repeat the problem, when I disconnect WiFi and print “DISCONNECTED,0,201 ”events, TCP can be disconnected normally

Patrik-Berglund, 2019-11-15:
Ok, this problem is not so often seen, could take a couple of days before it happens.

We have 10 devices in a test bench and in rare occasions after the disconnect event has been raised we can see that our SW still successfully manages to open a socket and run TCP traffic.

freakyxue, 2019-11-15:
@Patrik-Berglund   

I think in the case of WiFi disconnection, recv function will return an error, and TCP connection should be disconnected. Why does it run when WiFi is disconnected ，Need to analyze it with your code。
Can you write a demo to reproduce it and send it to me?

Alvin1Zhang, 2020-01-15:
@Patrik-Berglund Thanks for reporting, would you please help share if any updates for the issue? Thanks.

Patrik-Berglund, 2020-01-15:
Hi,

We have updated to the 3.3.1 SDK and are re-testing with 10 boards. Started the test on 2020-01-10, and will let it run for two weeks.

Alvin1Zhang, 2020-01-15:
@Patrik-Berglund Thanks for the updates. Looking forward to more feedback. Thanks.

freakyxue, 2020-01-20:
@Patrik-Berglund    Does the problem still exist

Patrik-Berglund, 2020-01-20:
@freakyxue 
Looks like it happened during this weekend but due to an error in the logging the serial output was lost. 

I understand if you wan't to close the issue and we can re-open it again when/if we have the logs.

poojavernekar, 2020-05-08:
Is it possible to have tcp/ip connection without WIFI and work by achieving internet gateway connection?

jack0c, 2022-05-20:
@poojavernekar Not understand what you want. ESP32 can use Wi-Fi or Ethernet to connect to local network for instant a router. You can use use PPPOE to connect to the internet directly. Do you want to use Ethernet interface or you want use PPPOE protocol? Anyway, I suggest you post a new issue, I will close this one.

closed 2022-05-20 by jack0c