espressif/esp-idf #4222: [BLE Mesh] esp_ble_mesh_provisioner_delete_dev does not reset device

Piill, 2019-10-21:
## Environment

- Development Kit:    ESP32-DevKitC v4
- IDF version: V4.0-beta1
- Build System: CMake
- Operating System:  Linux
- Power Supply: USB

## Problem Description

After having provisioned and configured a node, `esp_ble_mesh_provisioner_delete_dev` is called using the device UUID. I expect the device to be reset, and no longer part of the network. 

Instead, nothing happens. The device keeps functioning as usual.

Campou, 2019-10-22:
Hi @Piill 

Currently we recommend to send `Config Node Reset` message to reset a node, and for the API `esp_ble_mesh_provisioner_delete_dev`, we will add the reset funtionality into it.

Thanks.

Piill, 2019-10-22:
I never noticed that there was a node reset as part of the configuration client. I will use that instead.

Thank you.

Campou, 2019-10-22:
Hi @Piill 

Yes, you can send the configuration message with the opcode set to `ESP_BLE_MESH_MODEL_OP_NODE_RESET`. Besides the opcode, no other parameters will be needed.

Thanks.

Piill, 2019-11-01:
Hey @Campou 
i have a quick followup question.
When resetting a node, it never returns the "Config Node Reset Status".
Am i suppose to send that manually before resetting the node?

Campou, 2019-11-01:
Hi @Piill 

For resetting the node, there is no need to send the corresponding status message manually.
Thanks for the feedback, we are looking into this issue.

Thanks.

Campou, 2019-11-07:
Hi @Piill 

We have found the root cause of this issue, will resolve it ASAP.

Thanks.

Campou, 2020-07-27:
Hi @Piill 

In the latest idf, you can send Config Node Reset message without the need to set parameters (since no parameter is included in the message), please have a try.

Thanks.

Alvin1Zhang, 2020-07-27:
@Piill Thanks for reporting, sorry for slow turnaround. Feel free to reopen if the issue still happens. Thanks.

closed 2020-07-27 by Alvin1Zhang