espressif/esp-idf #4223: Bug in esp_wifi_set_country and esp_wifi_set_max_tx_power effectively disallows usage of the ESP32-WROOM-32 module in many countries

bkalytta, 2019-10-21:
### Environment:
ESP-IDF 4.1 (Latest) (as per version.mk)

### Issue

Concerning SDK documentation esp_wifi_set_country configures country info. The struct wifi_country_t among others contains the max_tx_power field which is the point of interest here.

I would expect that calling  esp_wifi_get_country would fill all fields correctly, in my case it contains following data:

```
wifi_country_t c= {0};
esp_wifi_get_country(&c);

+------------------+-------+
| FIELD            | VALUE |
+------------------+-------+
| c.schan          | 1     |
+------------------+-------+
| c.nchan          | 13    |
+------------------+-------+
| c.max_tx_power   | 2     |
+------------------+-------+
| c.code           | CN\0  |
+------------------+-------+
```
**First**, max_tx_power contains an invalid value (2). Concerning to documentation about `esp_wifi_get_max_tx_power` / `esp_wifi_set_max_tx_power`  :

> Maximum WiFi transmitting power, unit is 0.25dBm, range is [40, 82] corresponding to 10dBm - 20.5dBm here.

So the value of 0.5dBm would be invalid and also would not make any sense as default value.

**Second**, setting that value by calling `esp_wifi_set_country` has no effect at all. It was tried with all different values  like 0, 40, 82. `esp_wifi_get_max_tx_power` always returned 78 (78*0.25dBm = 19.5dBm).

Concerning documentation for struct `wifi_country_t` the field `max_tx_power` is described as 

> This field is used for getting WiFi maximum transmitting power, call esp_wifi_set_max_tx_power to set the maximum transmitting power.

So we could assume that this field has no effect on `esp_wifi_set_country` call and that we have to explicitly call `esp_wifi_set_max_tx_power` before or afterwards. Fine, the problem however is, that `esp_wifi_start` have to be called in before otherwise `esp_wifi_set_max_tx_power` will fail with error code `ESP_ERR_WIFI_NOT_START`.

### This means, we have to start the AP with invalid (forbidden) maximum transmit power configuration before we can change the transmit tx power.

### Expected behaviour

- `esp_wifi_get_country` would fill all fields with valid values
- `esp_wifi_set_country` would also set maximum transmit power
- `esp_wifi_set_max_tx_power` can be called before `esp_wifi_start`

esp-daiwei, 2019-10-21:
HI @bkalytta，thanks for your suggestions and totally agree with all your suggestions. We will fix/optimize the API accordingly.

Rhulin, 2019-11-05:
HI @bkalytta, Respond to your three questions:
**Question 1.** Currently we have not reproduced, I hope you provide a specific commit for us to reproduce. (Our test result is `max_tx_power` is 20dBm by default)
**Question 2.** The current version we have not used `esp_wifi_set_country` to set `max_tx_power`. If you set the `max_tx_power` value it is recommended that you use `esp_wifi_set_max_tx_power`. We will optimize this API in subsequent new releases.
**Question 3.** The problem of using `esp_wifi_set_max_tx_power` for `esp_wifi_start` is now resolved, and we will release it after the test is passed.
thank you so much for your suggestion.

bkalytta, 2019-11-05:
> **Question 1.** Currently we have not reproduced, I hope you provide a specific commit for us to reproduce. (Our test result is `max_tx_power` is 20dBm by default)

I think you're referring to that `c.max_tx_power` after calling `esp_wifi_get_country(&c);` has the value "2"? You can not reproduce it? So if you say the default value is 20dBm you should read a value of "80" here, is that right? If so, may be this was already fixed?

> **Question 2.** [...] We will optimize this API in subsequent new releases.

OK that sounds reasonable.

esp-HarveyRong, 2019-11-05:
HI @bkalytta  
By default, if you use `esp_wifi_get_country` to read the value of `c.max_tx_power` should be 20dBm, use `esp_wifi_get_max_tx_power` to read the value of  `c.max_tx_power` should be 78, the difference between the two is that the unit is different, that is the value read with `esp_wifi_get_country` is 0.25 *  `esp_wifi_get_max_tx_power`  read value.

bkalytta, 2019-11-05:
> HI @bkalytta
> By default, if you use `esp_wifi_get_country` to read the value of `c.max_tx_power` should be 20dBm, use `esp_wifi_get_max_tx_power` to read the value of `c.max_tx_power` should be 78, [...]

**If the units are different it should be documented somewhere.** But even then my returned value of "2" is not plausible since 0.25 * `esp_wifi_get_max_tx_power` should be 19.5 (19) but not 2? What value does `esp_wifi_get_country` return for your `max_tx_power`?

esp-HarveyRong, 2019-11-05:
> > HI @bkalytta
> > By default, if you use `esp_wifi_get_country` to read the value of `c.max_tx_power` should be 20dBm, use `esp_wifi_get_max_tx_power` to read the value of `c.max_tx_power` should be 78, the difference between the two is that the unit is different, that is the value read with `esp_wifi_get_country` is 0.25 * `esp_wifi_get_max_tx_power` read value.
> 
> **If the units are different it should be documented somewhere.** But even then my returned value of "2" is not plausible since 0.25 * `esp_wifi_get_max_tx_power` should be 19.5 (19) but not 2? What value does `esp_wifi_get_country` return for your `max_tx_power`? Is the value returned different after calling `esp_wifi_start` may be?

I hope that you can provide a specific commit ID, which will help me reproduce the `return value "2"` problem and debugging.

esp-HarveyRong, 2019-11-06:
Hi @bkalytta 
The `return value "2"` has been reproduced because the PHY has not been initialized before `wifi_start`. This value is unreliable if you use `esp_wifi_get_country` to get `max_tx_power` at this time.

bkalytta, 2019-11-06:
In that case please be sure that either `esp_wifi_set_max_tx_power` can be called **before** `esp_wifi_start` or that `esp_wifi_get_country` also set the max tx power and that it can be called **before** `esp_wifi_start`. Otherwise we could encounter the situation where the device has to be started with forbidden values first before it can be changed.

esp-HarveyRong, 2019-11-07:
**In fact,** when wifi is started, as long as the PHY initialization is completed, `max_tx_power` is the default value, which is 20dBm. `esp_wifi_set_max_tx_power` is used to limit the maximum power when the device is running.

closed 2020-07-30 by Alvin1Zhang