espressif/esp-idf #4225: win7 install of esp-idf-tools-setup-2.1 never completes

jmulchin, 2019-10-21:
Installing esp-idf-tools-setup-2.1.exe file under administrator never finishes. The following message in the install screen:
Running command: powershell -ExecutionPolicy ByPass -File "C:\Users\jmulchin\.espressif\dist\tools_WD_excl.ps1" -AddExclPath "C:\Users\jmulchin\.espressif\*.exe"
is where it has huing now for over an hour and a half. Have to cancel install to get out of it.
I have uploaded the setup log file for your information. Also, this is the 3rd time I've tried this.
[Setup Log 2019-10-21 #004.txt](https://github.com/espressif/esp-idf/files/3753215/Setup.Log.2019-10-21.004.txt)
I would assume that because the setup never completes that the development tools are probably not usable yet>???
Looking forward to a solution.

Jerry

jmulchin, 2019-10-24:
So, no answer to this problem? Great Support --- NOT!

negativekelvin, 2019-10-24:
What happens if you try to run the command separately?
`powershell -ExecutionPolicy ByPass -File "C:\Users\jmulchin.espressif\dist\tools_WD_excl.ps1" -AddExclPath "C:\Users\jmulchin.espressif*.exe"`

jmulchin, 2019-10-25:
At this point I don't want to try anything that might cause the ESP-IDF to stop working. Event though it did appear to not finish the install, when I tried using the package I could compile an example program. Now I'm waiting for my boards to show up to see if the code will run. If I find that the install did not install correctly at a later date then I will try your suggestion. But not now.
Thanks for getting back to me.

igrr, 2019-10-25:
Will keep this open, as it seems to be a valid issue in the installer.

pacucha42, 2019-11-01:
Hi Jerry,
I apologize for my delay, I'll try to fix the issue asap now. Would you share your Windows OS version, please? I don't see it mentioned anywhere. Thank you

Martin

jmulchin, 2019-11-01:
Hi Martin,

As the title of the message states, Win 7.

pacucha42, 2019-11-01:
I know, I noticed too late :-) Thanks for patience anyway. I'm back to you asap

chegewara, 2019-11-01:
And again the same question, what version system do you have?
Is it 32 or 64 bit windows 7?

chegewara, 2019-11-01:
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html#standard-setup-of-toolchain-for-windows

```
Currently only 64-bit versions of Windows are supported. 32-bit Windows can use the Legacy GNU Make Build System.
```

pacucha42, 2019-11-02:
I guess Jerry uses 64bit Win7, however, it might be helpful if you provide exact OS version as given by
**systeminfo | findstr /B /C:"OS Name" /C:"OS Version"** launched in the command line. Thank you

Stoi723, 2019-11-28:
Have same issue on windows 7 64 bit= Please help :(

pacucha42, 2019-11-29:
The fix is ready to merge (verified on both W7_64 and W10_64), GH code update coming soon. I apologize for the inconvenience, pls, hold on.
However, the remedy is trivial  even with the current code - simply uncheck the Windows Defender option and you should be fine. All this WD-scan-exclusions feature is just an optimization which usually increases the compilation speed, but it's not required for any ESP Tool operation by default -> the toolset is fully available, no features or functions are restricted without the ESP-IDF executables registered in the WD exclusion set

Hope this helps, at least for the meantime

Martin

jubeormk1, 2019-12-11:
> 
> 
> What happens if you try to run the command separately?
> `powershell -ExecutionPolicy ByPass -File "C:\Users\jmulchin.espressif\dist\tools_WD_excl.ps1" -AddExclPath "C:\Users\jmulchin.espressif*.exe"`

I tried that. I receive the next message in a win 7 machine x64 with security essentials:

```
PS C:\Windows\system32> powershell -ExecutionPolicy ByPass -File "C:\Users\boj\.espressif\dist\tools_WD_excl.ps1" -AddEx
clPath "C:\Users\boj\.espressif\*.exe"
Import-Module : The specified module 'Defender' was not loaded because no valid module file was found in any module dir
ectory.
At C:\Users\boj\.espressif\dist\tools_WD_excl.ps1:61 char:14
+ Import-Module <<<<  Defender
    + CategoryInfo          : ResourceUnavailable: (Defender:String) [Import-Module], FileNotFoundException
    + FullyQualifiedErrorId : Modules_ModuleNotFound,Microsoft.PowerShell.Commands.ImportModuleCommand

C:\Users\boj\.espressif\dist\tools_WD_excl.ps1 : Unable to find type [NotSupportedException ]: make sure that the assem
bly containing this type is loaded.
    + CategoryInfo          : NotSpecified: (:) [Write-Error], RuntimeException
    + FullyQualifiedErrorId : System.Management.Automation.RuntimeException,tools_WD_excl.ps1
```

After that in eclipse, following the guide provided [at Installing ESP-IDF Tools](https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md#installing-esp-idf-tools), when locating the installation directory I get this message: 
```
Executing C:\Users\boj\AppData\Local\Programs\Python\Python37\python.exe C:\Users\boj\esp-idf\tools\idf_tools.py export --format=key-value
C:\Users\boj\AppData\Local\Programs\Python\Python37\python.exe: can't open file 'C:\Users\boj\esp-idf\tools\idf_tools.py': [Errno 2] No such file or directory

```
I cannot start windows defender maybe because I am running microsoft security essentials.

I hope that helps, anyway windows 7 is about to close in January'20

Cheers.

If someone finds a workaround It would be very welcome...

pacucha42, 2019-12-13:
Hi jubeormk1,
based on the output provided:

1) running the PS script manually has the same effect as when launched from the installer process but appaerently, the Defender module is missing on your system. The PS Defender came preinstalled in Windows 8.1 and higher, you have upgrade your OS manually (via std Windows Update, PS cmdlets (Install-Package), or NuGet manager. Once you have the Defender module installed, verify you have got what you have been looking for:
**Get-Command -Module Defender**
You should see some 12 cmdlets, we are interested in  **Add-MpReference** & **Remove-MpReference** - if you got here, there is a fair chance the WD-exclusions part of the installation would be running well.

2) how did you launch the Python? I guess your environment vars are not propagated to other processes ... you usually need a fresh program start to get the block updated as the env. is created on a process startup. Restarting the Eclipse IDE might be trivial "fix" attempt, also check existence/contents of **IDF_PATH, PATH, OPENOCD_SCRIPTS** and **IDF_PYTHON_ENV_PATH** (required set) for the spawned(?) Python process - eg using Sysinternals ProcMon tool as it allows logging of filtered events.
Logfile might help as well, see **https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md#troubleshooting**

I assume you have reasons to run on Windows 7 so you either wait for our improved installer (close to public release, however, it doesn't install WD support to your OS) or expect some manual tweaking to set your environment properly. 

Let me know if you need more info
Martin

sapozhnikov, 2020-01-02:
Why it's closed? The problem still present in 2.2. Update minimal requirements or fix powershell script. (OMG, I've spent few hours trying to make it running, fixing scripts and restarting install.)
>you have reasons to run on Windows 7

Are you serious?

pacucha42, 2020-01-03:
> Why it's closed? The problem still present in 2.2. Update minimal requirements or fix powershell script. (OMG, I've spent few hours trying to make it running, fixing scripts and restarting install.)

Hi Dimitry,
it's unwanted side effect of the original issue fix - in your case we properly detect the Windows Defender module is missing but we disable wrong checkbox :-( Sorry about that, unfortunately this behavior appeared due to the last code update and was not caught in testing. This new issue is already fixed and shall be published asap - thanks for testing and I apologize for the troubles.

> 
> > you have reasons to run on Windows 7
> 

> Are you serious?
Well, Windows 7 support ends on January 14, 2020. That's about it

Martin

pacucha42, 2023-08-28:
This issue has been resolved long ago, closing the tracker. Feel free to reopen if needed

closed 2023-08-28 by pacucha42