espressif/esp-idf #4229: Updated ESP-IDF and get thrown a esp_http_client error

awilkie1, 2019-10-22:
Compiling .pio/build/esp32dev/esp_http_client/lib/http_utils.o
/Users/ashleywilkie/.platformio/packages/framework-espidf/components/esp_http_client/esp_http_client.c: In function 'esp_http_client_init':
/Users/ashleywilkie/.platformio/packages/framework-espidf/components/esp_http_client/esp_http_client.c:510:15: error: 'esp_http_client_config_t {aka const struct <anonymous>}' has no member named 'use_global_ca_store'
     if (config->use_global_ca_store == true) {
               ^

/Users/ashleywilkie/.platformio/packages/framework-espidf/components/esp_http_client/esp_http_client.c:516:15: error: 'esp_http_client_config_t {aka const struct <anonymous>}' has no member named 'client_cert_pem'
     if (config->client_cert_pem) {
               ^
/Users/ashleywilkie/.platformio/packages/framework-espidf/components/esp_http_client/esp_http_client.c:517:59: error: 'esp_http_client_config_t {aka const struct <anonymous>}' has no member named 'client_cert_pem'
         esp_transport_ssl_set_client_cert_data(ssl, config->client_cert_pem, strlen(config->client_cert_pem));
                                                           ^
/Users/ashleywilkie/.platformio/packages/framework-espidf/components/esp_http_client/esp_http_client.c:517:91: error: 'esp_http_client_config_t {aka const struct <anonymous>}' has no member named 'client_cert_pem'
         esp_transport_ssl_set_client_cert_data(ssl, config->client_cert_pem, strlen(config->client_cert_pem));
                                                                                           ^
/Users/ashleywilkie/.platformio/packages/framework-espidf/components/esp_http_client/esp_http_client.c:520:15: error: 'esp_http_client_config_t {aka const struct <anonymous>}' has no member named 'client_key_pem'
     if (config->client_key_pem) {
               ^
/Users/ashleywilkie/.platformio/packages/framework-espidf/components/esp_http_client/esp_http_client.c:521:58: error: 'esp_http_client_config_t {aka const struct <anonymous>}' has no member named 'client_key_pem'
         esp_transport_ssl_set_client_key_data(ssl, config->client_key_pem, strlen(config->client_key_pem));
                                                          ^
/Users/ashleywilkie/.platformio/packages/framework-espidf/components/esp_http_client/esp_http_client.c:521:89: error: 'esp_http_client_config_t {aka const struct <anonymous>}' has no member named 'client_key_pem'
         esp_transport_ssl_set_client_key_data(ssl, config->client_key_pem, strlen(config->client_key_pem));
                                                                                         ^
Compiling .pio/build/esp32dev/jsmn/src/jsmn.o
*** [.pio/build/esp32dev/esp_http_client/esp_http_client.o] Error 1

Alvin1Zhang, 2019-10-23:
@awilkie1 Thanks for reporting this. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

mahavirj, 2019-10-23:
@awilkie1 Did you also update git submodules? Can you please share output for `git submodule` command?

awilkie1, 2019-10-23:
So I did a platformIO update, which intern did an esp-idf update. 

this then added a new sdkconfig file to my project src file. it had a sdkconfig.h file already. Could this be conflicting ??

git submodule gives me no output.

mahavirj, 2019-10-23:
@awilkie1 We will need more information to comment or provide any help. I am not sure about which `ESP-IDF` version did PlatformIO upgrade to, but do you get same issue if you download latest `ESP-IDF v3.3` from either github repository or from release artifacts?

awilkie1, 2019-10-23:
@mahavirj we are using the IDF with the ADF framework. however, the IDF seemed to update and now has caused issues. 

the idf is now on 3.30300.0

awilkie1, 2019-10-24:
SOLVED. 

Issues arose due to the EDP-IDF not compatible with the ADF! need to put "platform = espressif32@1.9.0" in the platformio.ini file. 

Thanks for your help!

closed 2019-10-24 by mahavirj