espressif/esp-idf #4231: Failed to connect to ESP32 when flash

Tflgit, 2019-10-23:
The following problems occur when executing the command make flash：
App "blink" version: v3.3
Flashing binaries to serial port /dev/espDug (app at offset 0x10000)...
esptool.py v2.8-dev
Serial port /dev/espDug
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

I have successfully burned the example "hello_world", which is running normally. But I had this problem when I tried to burn another case. As I saw "connecting", I pressed and held the "boot" button, but it didn't seem to work. How can I solve this problem？

Alvin1Zhang, 2019-10-23:
@Tflgit Thanks for reporting this. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

Tflgit, 2019-10-23:
@Alvin1Zhang Thanks, The details:
hardware: ESP32_Lyrat
sdk configuration is as the default, modify the Serial port only

the board running "esp-idf/examples/get-started/hello_world" demo, version: v3.3
### board log
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6336
ho 0 tail 12 room 4
load:0x40078000,len:11624
ho 0 tail 12 room 4
load:0x40080400,len:6648
entry 0x40080764
I (32) boot: ESP-IDF v3.3 2nd stage bootloader
I (32) boot: compile time 09:30:05
I (32) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0812c ( 33068) map
I (107) esp_image: segment 1: paddr=0x00018154 vaddr=0x3ffb0000 size=0x01eb4 (  7860) load
I (110) esp_image: segment 2: paddr=0x0001a010 vaddr=0x40080000 size=0x00400 (  1024) load
I (116) esp_image: segment 3: paddr=0x0001a418 vaddr=0x40080400 size=0x05bf8 ( 23544) load
I (134) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x12b3c ( 76604) map
I (161) esp_image: segment 5: paddr=0x00032b5c vaddr=0x40085ff8 size=0x02058 (  8280) load
I (170) boot: Loaded app from partition at offset 0x10000
I (170) boot: Disabling RNG early entropy source...
I (171) cpu_start: Pro cpu up.
I (174) cpu_start: Application information:
I (179) cpu_start: Project name:     hello-world
I (184) cpu_start: App version:      1
I (189) cpu_start: Compile time:     Oct 22 2019 09:30:09
I (195) cpu_start: ELF file SHA256:  5ffe1057738350da...
I (201) cpu_start: ESP-IDF:          v3.3
I (206) cpu_start: Starting app cpu, entry point is 0x40081070
I (0) cpu_start: App cpu up.
I (216) heap_init: Initializing. RAM available for dynamic allocation:
I (223) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (229) heap_init: At 3FFB2EE0 len 0002D120 (180 KiB): DRAM
I (235) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (241) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (248) heap_init: At 40088050 len 00017FB0 (95 KiB): IRAM
I (254) cpu_start: Pro cpu start user code
I (272) cpu_start: Chip Revision: 1
W (273) cpu_start: Chip revision is higher than the one configured in menuconfig. Suggest to upgrade it.
I (276) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello world!
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in 4 seconds...
Restarting in 3 seconds...
Restarting in 2 seconds...
Restarting in 1 seconds...
Restarting in 0 seconds...
Restarting now.

burnig software: "esp-idf/examples/get-started/blink" demo, version: v3.3
### burning log
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /home/tfl/workspace/esp/esp-idf/requirements.txt are satisfied.

App "blink" version: v3.3
Flashing binaries to serial port /dev/espDug (app at offset 0x10000)...
esptool.py v2.8-dev
Serial port /dev/espDug
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
/home/tfl/workspace/esp/esp-idf/components/esptool_py/Makefile.projbuild:66: recipe for target 'flash' failed
make: *** [flash] Error 2

Tflgit, 2019-10-23:
@Alvin1Zhang I have solved this problem. To upload the binaries, the board should be put into upload mode. To do so, hold down **Boot** button, momentarily press **Reset** button and release the Boot button. Thanks!

Alvin1Zhang, 2019-10-23:
@Tflgit Thanks for the updates. Feel free to reopen if you still have the issue, thanks.

closed 2019-10-23 by Alvin1Zhang